opendata-stuttgart/sensors-software #145: Configurable ESP_ID

hakan42, 2017-09-26:
Wenn ein ESP ersetzt werden muß (zum Beispiel weil die USB-Buchse defekt ist) hat man keine gute Möglichkeit mehr, die bestehende Meßstation weiterzuführen. Die neue ESP reportet dann mit ihrer esp_id, und auf der Karte und in den Datenarchiven sieht es so aus als wäre an dieser Stelle ein neuer Sensor aufgetaucht und der alte weggestorben.

Momentan ist die einzige Lösung, eine eigene Firmware mit einem hardgecodeten String für die esp_id zu compilen und auto-update abzuschalten. Schön wäre es, wenn es ein Eingabefeld für die ID geben würde, die per default mit der hardware-id des chips befüllt wird.

Wenn diese Idee Anklang findet liefere ich gerne einen entsprechenden Patch.

ricki-z, 2017-09-26:
Oder man meldet sich bei uns und bittet um Änderung der ESP-ID. Dann wird die neue statt der alten eingetragen und alles läuft weiter. Im Archiv und auf der Karte werden dann weiterhin die gleichen API-IDs verwendet.

hakan42, 2017-09-26:
Oh, o.k., dann lese ich bei der nächsten Gelegenheit die "echte" ESP-ID des Chips aus und bitte um Anpassung. Ist vermutlich eine größe Fehlerquelle weniger...

dokape, 2017-09-26:
So eine Konfigurationsmöglichkeit wäre prinzipiell durchaus sinnvoll.

Zum einen wird beim bisherigen Vorgehen wieder auf Mails und persönliche Zeit Dritter (Entwickler?) Einfluss genommen. 
Zum anderen wird beim lokalen Speichern per (z.b. dem simple-data.php-script) die ID auch mit gespeichert. Änderung vom Sensor -> neue Datei-Namen. (ok, kann man per batch umbenennen oder script manuell anpassen)

Für den ersten Fall könnte natürlich auch eine Konfiguration durch ein personalisiertes Login auf Luftdaten.info (o.ä.) und Pflege/Zuordnung der ESP-ID zur 4-stelligen Sensor-ID später mal Arbeit abnehmen.

Adorfer, 2017-09-27:
I consider a manual process via email "to a person"  as general latent issue. A setup option or a self-service portal would be highly appreciated. 

If you really like to go for the "just write some mail" method: Please put in place a ticket systemstsm

dokape, 2017-09-28:
We have a crowd-based datagenerating.
But a one-person based Sensor-registering.   ;-)

My opinion: User should be able to register himself on a "luftdaten" or "arirohr" - Portal.
He should be able to create one or more "Airrohr"-"Box".
He should be able to assign the nodeMCU specific ID to that Box
(This is what one person does after the email right now)
One or more Airrohr-Boxes: because some persons manage more than one Airrohr (For friends, projects, family etc.)
further Milesstone: In his personal login, the user will find every airrohr specific information like his chart
(perhaps like on feinstaub.rexfue.de?) and should be able to download with one click his .csv files
I'm not really satisfied with the actual situation.
Some information on Luftdaten.info
some information on maps.luftdaten.info
some information at feinstaub.rexfue.de
some information at madavi.de
other information at github. 
The documentation is a part on Luftdaten.info and other parts at the Meta-Wiki on github.
specific documentation is directly in the readme at the sensor-software. 

All together does not look like a harmonic process. 
The project is discussed further more in media. On TV, in News, etc.
To get more people using this really easy project, IMHO the complete project should be reorganised based on the needs of noobs ;-)

Make it easyer to register and user can see at once the results. He will be satisfied in high grade.

ricki-z, 2017-09-28:
Ticket system is work in progress, also the self-registering of needed information for the sensors sensors. But we only have limited time. We can't do anything at once.
A configurable ESP ID could lead to a misuse of the system (more than one box with one ID sending data from different locations). Also a self-registering of sensors is a problem. One could create a bot that is registering all possible ESP IDs. But a registration of email and ESP ID over the ticket system should solve this. So more than one person can create the needed entries.
Information:
for Noobs all needed information should be on Luftdaten.info
for Powerusers additional informations can be found on the meta wiki
and for firmware developers there is more information in some files so that they don't have to switch between editor and browser. The information in this files will go to the wiki if the version is stable.

dokape, 2017-09-28:
I know time is limited.
But we can discuss the future. The roadmap, the pro and contra. Just my thoughts.

Adorfer, 2017-09-29:
>One could create a bot that is registering all possible ESP IDs.

1) this is nothing that can not be undone quickly
2) what serious motivation could somebody have to do such a thing
3) some type throtteling could be easily implementented after the first abuse cases
4) reasoning about potential abuse which will hardly have any big effect should be reserved to certain politicians (from the law&order corner)