opendata-stuttgart/sensors-software #147: BME280 temperature values seems to be (much) higher then DHT22 values.

janzieg, 2017-10-13:
Follow up of https://github.com/opendata-stuttgart/sensors-software/issues/137:

BME280 temperature values always seems to be higher then those from DHT22 (or other temperature sensors like SHT31). 

One reason for this is the Adafruit driver and its default configuration (used in the Airrohr firmware):
When using the default configuration, temperature values are 1-2 degrees too high! (and dependent other values are "wrong").
You can also check: http://www.esp8266.com/viewtopic.php?f=13&t=8030&sid=5b736d896eb9758f6ae56ce5aefbaceb&start=28

Reason: BME280 has some self heating depending its sampling configuration mode.

This can be optimizied by setting "oversampling" parameters during BME280 init (refer to spec https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BME280_DS001-11.pdf - "weather monitoring").

The Adafruit driver can enable this by calling "setSampling()" for "weather monitoring" mode:

`		// recommended settings for weather monitoring (see examples/advancedsettings.ino)
		// -> default settings sampling rate is quite high which heatens the sensor (temperature 0.8 C too high!)
		bme280.setSampling(Adafruit_BME280::MODE_FORCED, Adafruit_BME280::SAMPLING_X1, Adafruit_BME280::SAMPLING_X1,
						   Adafruit_BME280::SAMPLING_X1, Adafruit_BME280::FILTER_OFF);
`

This should be integrated into Airrohr-firmware - afterwards the values will be lower (and more exact).

mach0, 2019-02-04:
As I've got a BME 280 and a DHT22 inside .. This still seems to be not fixed or integrated .. is that true?

DeeKey, 2019-02-09:
Seems implemented. Here is the code:
https://github.com/opendata-stuttgart/sensors-software/blob/619ef6fab6a611d838bd7a71a29ce16f204b7eef/airrohr-firmware/airrohr-firmware.ino#L3551

@ricki-z please close this issue if it is so.

ricki-z, 2019-02-09:
This is implemented in the actual code.

closed 2019-02-09 by ricki-z