opendata-stuttgart/sensors-software #148: Request: rain sensor

scheich, 2017-10-16:
Hi,

what about implementing a rain sensor? I found this "YL-38 FC-37":

http://www.ebay.com/itm/Regensensor-Feuchtigkeit-Spritzwasser-YL-38-FC-37-fur-Arduino-Raspberry-Pi-DIY-/181797903638?hash=item2a53ffdd16:g:ZmYAAOSwuTxWCq20

Would be nice to report raining to own server and trigger a kind of alert(e.g. email): perhaps very helpful, if you dry your clothes outside. :-)

greetz
scheich

dokape, 2017-10-16:
Nice. But what does it messure? Liter/square-meter? 
Or just: light wet, wet, very wet? 
We have the measure of humidity with several sensors.

Adorfer, 2017-10-16:
both are just "raining yes/no"-sensors. 

the YL38FC37 is working based on electrical resistance. 
the RG11 works on optical effects on the sensor surface. 

both are unsuited for detecting volume of rain. 
both are heavily dependend on surface-structure. Any dirt (algues, bird feeces, spider webs) will have heavy influence and potentially delay reactions of sensors (especially "at the end of rain", if the dirt behaves like a sponge....
So at least bi-monthly cleaning is highly recommended.

it's not that volumetric sensors do not need regular cleaning, but there it makes at least sense to go into the effort. 

a cheap solution might be to hack into something like 
[image]

dadosch, 2017-10-26:
There is already quite a cheap optical solution with a acceptable (but still not as precise as tip bucket solutions) available called  ["RG-11".](http://rainsensors.com/rain-gauge-modes/)

Adorfer, 2017-10-29:
@scheich you were actually right with your suggestion of Hydreon RG-11. 
It actually does -from what i gathered now- a rather decent job in "bucket mode", counting individual rain drops in number an size. And cleaning is -compared to dip-buckets devices- rather easy to do and in plus less frequent neccesary. 

(the reference i looked up were setups for thereshold based switching of irrigation systems etc, not real weather monitoring systems.)

The issue i have with the RG11 is: Where to get one?

http://shop.weatherstations.co.uk/hydreon-rg-11-optical-rain-sensor-800-p.asp
states a price of around 100€, which looks rather advanced for mey, or at least taking into acount my ambition in fitting this to sensors which are currently not "free from all sides", meaning that some building behind is shielding them, rendering the output rather tainted, depending on the direction and strength of wind. 
[image]

scheich, 2017-10-29:
yes, thats expensive. On http://rainsensors.com it would be 60$ + shipping/tax.

Adorfer, 2017-10-29:
> On http://rainsensors.com it would be 60$ + shipping/tax.
30$ shipping, and due to the offered transport most likely unavoiable import tax (19%), and with a bit of luck no customs fees (unless you fall into the famous EMS pithole)

scheich, 2017-10-29:
yes, uk store is the better choice and probably the faster one too

jmberg, 2019-01-06:
FWIW, I started doing some work on a simple one:

https://github.com/opendata-stuttgart/sensors-software/pull/357

lichtteil, 2019-06-16:
What happend to this topic? @scheich how did you solve this? I would also like to detect rain (most important when it starts to rain) by using a 50 cent PCB that just goes from LOW to HIGH when there are water drops on the surface.
I have no idea what is the best way to integrate it with the sensor firmware. Do I need to code it myself?

ricki-z, 2019-06-19:
We won't implement this in this firmware. We try to measure air pollution. For a weather station there may be better projects than this.
(The memory of the NodeMCU is very limited. Thats why we try to focus on our main topic)
So yes, you need to code this yourself if you want to use a rain sensor with the same device.

lichtteil, 2019-06-27:
@ricki-z don't worry, I fully understand that :) I thought that @scheich maybe found a good solution I guess you can close the Issue then