opendata-stuttgart/sensors-software #150: Build with updated Expressiv-Libraries for KNACK (WPA2/ieee80211r etc)

Adorfer, 2017-10-18:
Just to document it here: I assume that the upcoming (binary) releases will be made with the updated Libraries for Arduino/ESP8266.
https://github.com/esp8266/Arduino/issues/3725

dokape, 2017-10-18:
Arduino lib updated

https://github.com/esp8266/Arduino/issues/3725

ricki-z, 2017-10-18:
The updated version is 2.4.0-rc2 only. Seems we need to wait for the official release of version 2.4.

dokape, 2017-10-19:
AS I'm not a developer and not very familiar with the ArduinoIDE, I'm not shure which libraries and where and how they have to be updated in arduino to Compile and flash. :-)

dokape, 2017-11-05:
Is the new ESP8266-Librarie (KRACK)  just used for compiling the source? '

ricki-z, 2017-11-05:
And again: latest ESP8266 library with KRACK patches is RC1 (release candidate)
'Normal' development steps:
Alpha: including new features and fixes, may be very unstable
Beta: mostly feature complete, for testing and bug fixing
Release candidate: most bugs removed, should be ready for release, but there may be bugs, that weren't found until now.
Release: the finished and published version

reibuehl, 2017-11-05:
I thought that RC2 is the one with the fixes??

ricki-z, 2017-11-05:
Latest ESP8266 library with KRACK patches is RC2 (release candidate)
Please look at: https://github.com/esp8266/Arduino/issues?q=is%3Aopen+is%3Aissue+label%3A%22type%3A+bug%22+milestone%3A2.4.0
Many open issues for this version ...

'Normal' development steps:
Alpha: including new features and fixes, may be very unstable
Beta: mostly feature complete, for testing and bug fixing
Release candidate: most bugs removed, should be ready for release, but there may be bugs, that weren't found until now.
Release: the finished and published version

Adorfer, 2017-11-06:
so I understand from this statement: "an rc is not a stable release, so we will not use it for a stable airrohr either". Correct?
(if there is a different condition, I might just not have understood the reply and missed something. )

ricki-z, 2017-11-12:
this is correct.

ricki-z, 2017-11-19:
First test with esp8266 Arduino lib 2.4.0-RC2:
only 2% available space left in flash mem. This means around 16000 Bytes for certs and other improvements.

tux1337, 2018-01-06:
Arduino core for ESP8266 is now in version 2.4 released (final) 

I used the esptool from 2.4 to flash NRZ-2017-099, is this anything that I have to do to ensure that the sensor use now the new library with the included KRACK patch? 

If yes: I can confirm that everything is working with 2.4

ricki-z, 2018-01-10:
This will only flash the unmodified firmware. You need to completely recompile the source code.

ricki-z, 2018-01-17:
Its not possible to compile a working firmware with version 2.4. Compilation with this version will end up with a non-working web config.

HarmOtten, 2018-01-22:
When can we expect an update of the sensor-software that is compatible with ESP8266 v2.4.0?

ricki-z, 2019-01-23:
Actual version is compatible with ESP8266 v2.4.0. Closing this issue.

closed 2019-01-23 by ricki-z