opendata-stuttgart/sensors-software #151: Request: Auto Update for beta versions

nobels, 2017-10-22:
Today i had the time to upload the beta firmware.

So i have to connect my device with USB to my Computer.
Manual upload the bin file. 

But i had this idea that we could check auto-update to latest version with (Beta-versions)

Would this be nice ?

So you would have

Version 100-BETA
Version 99 (Latest version)
Version 98

ErikLeysen, 2017-10-22:
Sounds fine form me

ricki-z, 2017-10-23:
The auto update for beta versions will be available in the next version, see beta branch for this.
Then you can configure the use of beta versions.

dokape, 2017-11-25:
NRZ-2017-100-B9/10

Option is implemented, but has still error.
Produces conflict with Update (more downgrade) to stable Version.  See #169

ricki-z, 2018-08-09:
@nobels Should work with new actual release NRZ-2018-105. Can we close this issue?

closed 2019-01-23 by ricki-z