opendata-stuttgart/sensors-software #152: NEO 6m Issues

dokape, 2017-10-28:
Software: NRZ-2017-100 beta 4

Doku falsch:  RX und TX-Pin in Doku vertauscht. 
Falsch: RX-Pin vom NEO -> PIN D5
Falsch: TX-Pin vom NEO -> PIN D6

Richtig: RX-Pin vom NEO -> PIN D6
Richtig: TX-Pin vom NEO -> PIN D5
Bitte bestätigen! 

values-Seite: GPS-Werte werden nicht angezeigt. (ist gar nicht vorgesehen. keine Kontrollmöglichkeit)

Display: Keine Anzeige der GPS-Werte. Keine Kontrollmöglichkeit. 
Insbesondere in Hinblick auf mobile Verwendung wäre eine Kontrolle am Display sinnvoll.

In den /data.json Werten sind die Daten enthalten.

Adorfer, 2017-10-29:
the RxD/TxD discussion is frequent one, since -like on every null-modem-link- the one unit's TxD has to go the the other unit's RxD and vice versa. 
So you may technically correct label the pins according to their function on the respective PCB as TxD where the bits come falling out. And ask the users to make the cross between RxD and TxD with their wiring. 
Or you label/name them that you connect 1:1 "by the name" making it technically incorrect for (mostly) the "smaller" PCB. 
And you have the situation (which i assume here) where somebody made the naming "for 1:1 connection" and the user (=the person making the documentation) did the crossover anyway (a second time), rendering the setup disfunctional. 

anyway: It has to be done in a way here, a normal user can wire it up without contamplating about technical background. So your suggestion is right.

dokape, 2017-10-29:
So PIN D5 is RX and D6 Is TX. Correct.
And RX Neo has to go to TX nodeMCU
And TX Neo has to go to RX nodeMCU

The documention is wrong. Because it describes the wiring. And describes TX to TX and RX to RX.
I will fix it.

dokape, 2017-10-30:
Documentation changed. 
Pull-Request done. 

2do: 
values-page to show GPS-Data
Displays to show GPS-Data

ricki-z, 2017-11-01:
values page is updated in beta version B5.

dokape, 2017-11-06:
GPS-Data are on values-page.

GPS-Data are shown before first messurements. Thats ok, but why show GPS-DATA before all other data? 
In that case, BME, HTU, DHT-Data could also be shown...

By refreshing the values page several times I got sometimes the values for Date and Time double as in picture below. Used Chrome 61.0.3163.100 on Windows 10.

GPS-DATA not shown on Displays?

[image]

dokape, 2017-11-06:
GPS-DATA are not stored at the Data-simple.php code provided from the madavi-api.
I Thought, thats a "catch-all"?

ricki-z, 2017-11-07:
Showing GPS data before first measurement:
We need to read the GPS data as soon as possible. The SoftwareSerial uses a buffer for incoming data. And we had some crashes if this buffer filled up.

Double date/time:
Will be corrected in version -B7

dokape, 2017-11-07:
Ok. "GPS-Data at once" should be documentated to technical reasons.

dokape, 2017-11-11:
NRZ-2017-100-B7
Double shown date and time fixed. > Looks good.

Data on OLED:  the  . . o . . 
is shown double. 
see also here: https://github.com/opendata-stuttgart/sensors-software/issues/143#issuecomment-343656641

dokape, 2017-11-13:
OLED:  the "navigation" . . o . . is now shown correct.

The GPS-Values are shown correct on web-page

for DOCU:
The GPS-Values are shown on OLED and LCD and web.
LCD: only LAT and LON are shown due to only 2x16 letters/numbers. 
OLED: Alt is shown in addition to LAT and LON
Webpage: shows additional DATE and TIME of GPS.

Issue is solved. 
will be closed when productive.

DeeKey, 2019-12-30:
@dokape @ricki-z is it still an issue or can be closed?

ricki-z, 2019-12-30:
This is still an issue. Some users try this with an ESP8266 and this won't work in most cases. With support for ESP32 this may change.

DeeKey, 2019-12-31:
Them support for NEO 6m must be marked experimental if it is not working reliably.

ricki-z, 2020-01-01:
It is marked as experimental.

DeeKey, 2020-01-01:
Well, no. On the latest beta 128 B5 in config I see no "Experimental" tag for GPS (NEO 6M) option...