opendata-stuttgart/sensors-software #153: No json-Data until first messurement

dokape, 2017-10-28:
the json-data at #Sensor-IP#/data.json 
is not available until first messurement.

To prevent parsing-errors due to not existing data I would suggest to provide a json-data-array without values instead of no data.

ricki-z, 2017-11-02:
New beta should have data.json without sensor data, but with version and tim to first reading.

dokape, 2017-11-05:
NRZ-2017-100_BETA5:

Donwload source as zip from beta branch.
compiling with arduino as last betas.
runs through.
uploading: no error.

starting software leeds to reboot:
------- outut serial console ----------

mounting FS...
mounted file system...
reading config file...
opened config file...
parsed json...
Starting Webserver... 0.0.0.0
output debug text to display...
6
Connecting to Sense
........
WiFi connected
IP address: 192.168.200.52

ChipId: 11729146
Read SDS...
Read DHT...
Read HTU21D...
Read BME280...
Read GPS...
Show on OLED...
Show on LCD 1602...
Trying BME280 sensor on 76 ... found
Stopping SDS011...

Exception (3):
epc1=0x40224da6 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40263314 depc=0x00000000

ctx: cont 
sp: 3fff3560 end: 3fff3790 offset: 01a0

>>>stack>>>
3fff3700:  3fff267c 000000c8 000000c8 4010020c  
3fff3710:  3fff2392 3fff2390 3fff3740 3fff2394  
3fff3720:  3fff2392 3fff2390 3fff2684 40207bac  
3fff3730:  3fff2550 00002580 3fff2684 40213498  
3fff3740:  00000000 00000000 00000000 00000000  
3fff3750:  00000000 00000000 feefeffe feefeffe  
3fff3760:  feefeffe feefeffe feefeffe 3fff2760  
3fff3770:  3fffdad0 00000000 3fff2758 4022344c  
3fff3780:  feefeffe feefeffe 3fff2770 40100718  
<<<stack<<<
H!⸮⸮⸮⸮⸮⸮⸮

dokape, 2017-11-06:
NRZ-2017-100-B6:

data.json generated like this:

I suggest to display the sensor-ID

`
{"software_version": "NRZ-2017-100-B6", "age":"89", "sensordatavalues":[]}
`

ricki-z, 2017-11-07:
sensor-ID is send as http header. I think the json should be mostly the same as the json with values.

dokape, 2017-11-07:
ok.
Then we can close this issue when productive.

closed 2018-08-21 by ricki-z