opendata-stuttgart/sensors-software #160: NRZ-2017-99: no web, not sending data, but pingable+dhclient alive

Adorfer, 2017-11-06:
i find a unit which from time to time does not send data. 

with a further look i discovered that
- it is responding to ping reliably
- it is talking to the dhcp-server (req/ack of leases)
- it is not taking connects on Port80 (connection refused)
- no data to madavi, nor opensensors. 

If i cut the wifi-uplink, then look like it's rebooting after around 10 minutes and is back to normal operation. 

And sorry, i do not have any chance to observe this on the serial console, since the device is already sitting in a location where i only have acces with long ladder and need the person present who arranged the location...
So not much i can contribute, i want to raise awareness, that such an situation may occurr and that it can be "cured" by shutting down the wifi uplink for a few minutes. 

https://www.madavi.de/sensor/graph.php?sensor=esp8266-555672-pms

dokape, 2017-11-06:
try to deactivate the opensensemap. I had this issue too.

Adorfer, 2017-11-06:
> try to deactivate the opensensemap. 

deactiviation of opensensemap is not an option here, since the sensor is using an PMS5003 and it has been decided that PMS will not be displayed on the "luftdaten.info" map.

Adorfer, 2017-11-06:
i deduct from #144 (which seems to discuss more about different wifi-uplink problems), that there might be a race condition during http-push to opensensmap-api. 
Nevertheless, if (this is speculative) this curl/wget/whatever get's stuck during comm due to network/wifi issues, then the result should ne be "dead main loop" of sensor, but some kind of timeout or watchdog should reset the unit as a last resort if it's stuck for several minutes. 

(this is really just speculative, i do not have any proof, so please do not take my recommendation for granted, real traceback is needed.)

dokape, 2017-11-06:
In #144 i had 2 different problems. First is same as you describe. No data, but pingeable. Then reboot of Sensor and works for a time between 10min to 2hours and stucks again. 
I had this problem also with a testsensor in the serial console and there I saw the stuck while sending data to opensensemap. I deactivated opensensemap and Sensor runs again perfect. 
Two/three days later I updated the Router and got network issues which i first thought they were based in the first problem. But this wasn't.

Try to deactivate for testing opensensemap to see if your problem still exists. 
If it runs, we have an issue with the firmware and opensensemap API.

ricki-z, 2017-11-12:
Sometimes the SSL connection to opensensemap needs too much memory (RAM). We try to solve this by reducing the response headers.

dokape, 2017-11-25:
It's still an issue? Could you solve the problem? Any new information about your problem?

offlinehoster, 2018-02-17:
Is this still a problem? Today i connected my new DHT22. I would like to check if he is working. And i'm not able to access the web interface. Any reasons?

I'm not using opensense map. Just luftdaten.info.

ricki-z, 2018-02-18:
How did you try to access the web interface?
You could use http://www.madavi.de/sensor/graph.php to look for the DHT22. Search for your esp8266 ID. You should find entries for the SDS011 and the DHT.

dirkmueller, 2019-11-12:
@Adorfer still an issue? can this be closed?

Adorfer, 2019-11-20:
the sensor was lost quite a lot updates ago. i do not know if it still has power supply. 
issue can be closed.

closed 2019-11-20 by Adorfer