opendata-stuttgart/sensors-software #162: NRZ-2017-100-B6 after some mesurements no PM-Data is shown

dokape, 2017-11-06:
NRZ-2017-100-B6

After 2 to 4 mesurements run through fine, the PM-Data get lost. 
At value-Page nothing is shown.
On Display nothing is shown,
The data.json does not contain the values any more (in the runs before it was there!)

[image]

data.json:
`{"software_version": "NRZ-2017-100-B6", "age":"64", "sensordatavalues":[{"value_type":"temperature","value":"5.60"},{"value_type":"humidity","value":"65.60"},{"value_type":"BME280_temperature","value":"5.78"},{"value_type":"BME280_humidity","value":"74.23"},{"value_type":"BME280_pressure","value":"97378.70"},{"value_type":"GPS_lat","value":"47.xxx315"},{"value_type":"GPS_lon","value":"8.xxx872"},{"value_type":"GPS_height","value":"47.74"},{"value_type":"GPS_date","value":"11/06/2017"},{"value_type":"GPS_time","value":"19:52:45.00"},{"value_type":"samples","value":"223224"},{"value_type":"min_micro","value":"263"},{"value_type":"max_micro","value":"1730019"},{"value_type":"signal","value":"-51"}]}`

dokape, 2017-11-11:
NRZ-2017-100-B7

Calling the /data.json in Browser,
the next cycle / run PM-Data will not be displayed on Displays, in Web and will not be transferred to influxx, or own API.

ricki-z, 2017-11-12:
On my node I can't replicate this. I tried this some times but every time the values were shown as expected on both the displays and the values page.

dokape, 2017-11-12:
After doing More tests and changing setup:
I'm mit shure if calling the data.json data is the trigger.

Letting the Sensor running for some hours, a lot of SDS and DHT values are not logged/shown/transmitted
To my custom API. 
Going back to NRZ-2017-100-B4 solves the problem without changing hardware setup or config. 
So I might suggest it is based in the changes between -100-b4 and -100-b6

dokape, 2017-11-25:
NRZ-2017-100-B10
seems to be stable. Issue cannot be replicated yet.

ricki-z, 2018-06-09:
@dokape can we close this issue?

ricki-z, 2018-08-09:
@dokape Should work with new actual release NRZ-2018-105. Can we close this issue?

ricki-z, 2019-03-14:
Closed after 6 months of inactivity.

closed 2019-03-14 by ricki-z