opendata-stuttgart/sensors-software #166: License is missing

pathmapper, 2017-11-12:
For software in this repository and also for the data. Related: https://github.com/opendata-stuttgart/sensors-software/pull/165

ricki-z, 2017-11-14:
License for directory airrohr-firmware is GPL v3. License.md was added to master branch. Beta branch should also contain License.md. Other parts of this repo may have other licenses (i.e. copies of libraries, ...).
License for data (Database Contents License (DbCL) v1.0) will be added to the website in the next few days.

pathmapper, 2017-11-15:
Thanks @ricki-z, it is already online:
[image]

- What about adding the license information also to the [readme.md](https://github.com/opendata-stuttgart/sensors-software/blob/master/readme.md) of this repository saying it is valid for the airrohr-firmware directory? Currently the license is a little bit hidden in the directory.

- What about a license for the [feinstaub-map repository](https://github.com/opendata-stuttgart/feinstaub-map): https://github.com/opendata-stuttgart/feinstaub-map/issues/52 ?

- Just for my understanding: Am I right that the [ODbL ](https://opendatacommons.org/licenses/odbl/1.0/)is also applicable, because of [DbCL](https://opendatacommons.org/licenses/dbcl/1.0/)

> 2.2 Conditions of Use. You must comply with the ODbL.

If yes, what is the reason for choosing DbCL instead of ODbL with DbCL?

ricki-z, 2017-11-15:
The license will be added to the INO file (and others like readme.md) in the next merge from beta to master. I would like to keep this workflow.
We are working on a new feinstaub-map. So we would like to publish the license in this version.
Difference between ODbL and DbCL:
ODbL - Open Database License -> Database only, not content
DbCL - Database Content License -> published content only, not database
For the database we can choose another license as long as this license complies with the ODbL. We have to read and understand all those licenses. And as we may have personal data in our database we can't open the whole database to the public. Our would you prefer to have your address being published?

pathmapper, 2017-11-15:
@ricki-z thanks for these information!

> We have to read and understand all those licenses. 

This is what I am trying to do:-) I was just wondering if there could be a "stand-alone" DbCL because it requests to comply with the ODbL. See point 2.2 of DbCL and also: https://opendatacommons.org/faq/licenses/#db-versus-contents

>  And as we may have personal data in our database we can't open the whole database to the public.

I was thinking about a separate database only for the published data without personal information (like the addresses you've mentioned).

ricki-z, 2017-11-15:
And for this we have to re-'design' the database before publishing a license

DeeKey, 2019-12-30:
@ricki-z if I am not mistaking then you have added the license?

ricki-z, 2019-12-30:
We need to check all directories and repositories. This is time consuming.

DeeKey, 2020-07-19:
License information page is missing on the new site https://sensor.community/

ricki-z, 2020-07-19:
@DeeKey this comment should be a new "issue" in the sensor.community repo (and not here). How should the maintainers of the site repo see this if you post this here?

closed 2020-07-19 by ricki-z