opendata-stuttgart/sensors-software #167: NRZ-2017-100-B9: AP-Mode, /config page: error_content_length_missmatch

dokape, 2017-11-14:
Testing the AP-Mode functionality.

Changed the WLAN by misspelling. -> connection to WLAN not possible.

forced the sensor to switch to AP-Mode: fine. works. 
On OLED, the AP-Mode is shown, with (customized) AP SSID-Name. Works.

Connect to the custom SSID: works

Call the AP-IP 192.168.4.1: 
redirect to /config-Page. Should be. Works.

But: Calling the /config-page results in 9 of 10 times in Error.
Browser Chrome says: error_content_length_missmatch

The config-page is quite not access
testet with: 
iphone oOS 11
Android 5 /cyanogenmod 12.1
Android 6.0.1 / cyanogenmod 13.1.2
Android 7.0
Windows 10, Chrome 62.

Can someone do a test against?

ricki-z, 2017-11-14:
This should use the same function calls to generate the pages as in normal mode. And there the content length is correct ...

dokape, 2017-11-14:
So therefore i ask someone with other setup/WLAN/ Smartphone/tablet to check against...

ricki-z, 2017-11-15:
I will test this today. I tried to get a long run over 24 hours. Until now this seems to work (SDS011, BME280, LCD, OLED). Until now 600 measurement cycles without crash.

dokape, 2017-11-15:
SDS011, BME280, LCD, OLED are stable here as well. Never had problems with that.

dokape, 2017-11-16:
Deleting complete filesystem, compiling complete NRZ-2017-100-B9 
starting up Sensor:
Problem with "error_content_length_missmatch" still exists.
Only DHT22, OLED, LCD and SDS011 connected. 
activated only DHT and SDS in vanilla config. 
had to reload the config page about 10 to 15 times until I got a working config-page.

ricki-z, 2017-11-16:
Version B10: Changed wifi SSID to something unknown. Node starts config mode.
Neither chrome nor firefox (both under MacOS High Sierra) show "error_content_length_mismatch".
The connected gps seems to need to much cpu cycles (interrupts). So sometimes no content is shown.

dokape, 2017-11-25:
NRZ-2017-100-B10

look better, but still happens. (No GPS connected and not activated)

I had this when I changed the standard AP-Name to something like "MeinSensor-117...."
Testet with an older Android-Tablet where no other WIFI SSD is stored.

ricki-z, 2018-06-09:
Could you test this with version NRZ-2018-103? May be we can close this issue.

closed 2019-04-01 by ricki-z