opendata-stuttgart/sensors-software #168: BasicAuth cannot login

azrdev, 2017-11-16:
I assembled a nodemcu v3-sensor, a friend who already did so multiple times flashed the firmware onto it, now I started to configure it. logged into the "Feinstaubsensor-ID" wifi, went to the config page, and first set BasicAuth for the admin page (since the sensor should run in freifunk). Clicked Save which triggered a reboot

Now, if I (re)start the sensor, its wifi AP comes up, and the (config) website asks for username + password, but I cannot log in with my configured credentials. Both have 24 chars in `a-zA-Z0-9` (generated by my password manager), which should not hit a length limit looking at the [code][1].

any other idea what might be wrong here (on my side or in the firmware)?

[1]: <https://github.com/opendata-stuttgart/sensors-software/blob/master/airrohr-firmware/airrohr-firmware.ino#L1092>

ricki-z, 2017-11-16:
Could you please try the default values?

dokape, 2017-11-16:
If the login is not possible, there is a way do delete the configuration.

https://github.com/opendata-stuttgart/meta/wiki/Konfiguration-der-Sensoren#l%C3%B6schen-der-konfiguration

You need the Arduino IDE.

azrdev, 2017-11-16:
@ricki-z `admin:feinstaub` does not login, neither do `admin:$mypw` nor `$myuser:feinstaub`

dokape, 2017-11-16:
NRZ-2017-100-B9

Setting a short username/Password works.

Setting a 30 chars Username/Passwort: Same problem as @azrdev reported.
(I used 123456789012345678901234567890 as PWD and Username....)

And now I can check how to delete the config.

ricki-z, 2017-11-16:
Could someone look at the code (airrohr-firmware AND ESP8266WebServer) if there is any limit beside the 64 chars I've found?

dokape, 2017-11-25:
I don't unterstand the code in that deep.

dokape, 2017-11-27:
Next step for testing would be to enlage one by one the user/pwd length until it stops working. 
The try the one smaller user/pwd if it is a problem with storing the PWD...

dokape, 2017-12-30:
NRZ-2017-100-B11

Testing today:
enlarge Username from 8 to 25 works. (12345678 to 1234567890123456789012345)
Then enlage PWD from 8 to 26 (12345678 to 12345678901234567890123456) also worked.

There there any changes done?