opendata-stuttgart/sensors-software #169: NRZ-2017-100-B9, downgrad  beta to master on activated Auto-Update

dokape, 2017-11-16:
Config deleted by uploading new SPIFFS
compile and upload NRZ-2017-100-B9.

Startup with AP.

select WLAN and provide PWD.

-> Reboot with connect to the WLAN.
-> Auto-Update will be forced
-> Auto-Update will downgrade to standard (older master) Firmware.

You are not able to disable the Auto-Upgrade earlier. 

Suggestion:

Provide a variable:

#define IS_BETA "B9"
#define SOFTWARE_VERSION "NRZ-2017-100" + IS_BETA

Later:
If IS_BETA <> "" then ....  AUTO-UPDATE to BETA-BRANCH (or no AUTO-UPDATE)

dokape, 2017-11-25:
NRZ-2017-100-B10

Option: 
[ ] Auto-Upgrade: does an Auto-Update to latest stable Version (even when newer Beta is installed)
[ ] Lade Beta Versionen :  does an Auto-Update to latest Beta-Version (could not test as I have latest Beta)

dokape, 2017-12-30:
NRZ-2017-100-B11

Beta-Firmware flashed on empty filesystem
Connected to WLAN
Autoupdate: downgrade from actual 100-B11 to 099 stable  :-/ 
Flash beta again...

ricki-z, 2017-12-30:
As mentioned more than one time, the feature 'use Beta version' is implemented the first time in this version. This is work in progress. You have to disable auto update if ou use beta versions.

ricki-z, 2018-07-11:
Sensors with 'use Beta version' enabled should now receive the latest beta version.

ricki-z, 2018-09-10:
@dokape the actual versions shouldn't go back to normal version. Can we close this issue?

ricki-z, 2018-09-17:
the version nr. of betas should always be euqal or greater than that of the release version.
Some users might be interested in testing the beta. But if they decide to change back because of the instabilities they should be able to go back to the release version. And I would expect that the sensor will go back to the last release version after disabling 'Beta versions'.
I will add an extendet check in the update script so that a manually flashed version won't get overwritten by auto update if the version is newer.

LuchtwachtersDelft, 2018-09-18:
Thanks, that would really help us out. 

We had a workshop last night with a manually flashed customized version of 110-B8 which worked pretty well with Android, iOS and Windows laptops. By the end of the workshop I saw 109 on someone's screen, imagine my horror. :scream:

I know, bad practice going live with a beta version, but I felt I had to do this because of a meeting of citizen science projects next Friday and some people have been waiting for months for the connection fix.

ricki-z, 2018-09-18:
Compiling is finished, new beta version is pushed to Github and to update server

LuchtwachtersDelft, 2018-09-18:
I think the last betas are a huge step up from 109.

I still want to test if something like this makes any difference, but maybe for 111.

https://github.com/LuchtwachtersDelft/sensors-software/commit/fea03b6f5cf3571e180c5015db63e4e9d7f282a0

ricki-z, 2018-09-18:
Version 111 would be the next release version, next beta is 112.
If the actual beta is solving most of the problems for the initial connect and config, I would like to make this the next release version 111.

ricki-z, 2019-11-13:
@dokape can we close this issue?

dirkmueller, 2019-11-27:
Please reopen if there is still something to fix. From reading over it, this seems resolved. Thanks!

closed 2019-11-27 by dirkmueller