opendata-stuttgart/sensors-software #170: NRZ-2017-099, Luftfeuchtigkeit zu hoch, meist 99,9%

msebald, 2017-11-22:
Ich stelle seit Monaten fest, dass die Luftfeuchtigkeit, die mein Sensor anzeigt, viel zu hoch ist - meist sehe ich ausschließlich 99,9% über Stunden, Tage, Wochen. Dies betrifft also nicht nur die aktuelle Firmware sondern scheint ein generelles Problem zu sein. Vor allem auch deswegen, weil ich dies an zwei weiteren Sensoren beobachten kann, die von zwei Freunden betrieben werden. Bei diesen drei Sensoren steht uns der Vergleich mit anderen Messgeräten zur Verfügung, die jeweils weit von den 99,9% entfernt sind. Ein Blick auf openSenseMap zeigt, dass ca. 2/3 der Sensoren bei 99,9% hängt.

Ist das Problem bekannt? Liegt es an der Hardware oder ist es ein Softwarethema.

Bisher hat mich diese Sache nicht gestört, da es mit dem eigentlichen Thema Feinstaub ja nichts zu tun hat. Ich verwende allerdings Temperatur (diese scheint soweit im Rahmen zu sein) und Luftfeuchtigkeit in openHAB und will damit bestimmte Dinge steuern.

Adorfer, 2017-11-22:
Die BME280 sind teilweise extrem helligkeits(!)-empfindlich. Schon eine hochfrequent flackernde Leuchtstofflampe (CFL) in mehreren Metern Abstand bringt die zum Absturz. Dann gibt es aber idR gar keine Werte mehr bis zu nächsten Powercylcle (das Geräte muss also Stromlos sein)
Wenn die BME280 jetzt aber das gleiche Problem bekommen wie die alten DHT22ern, die nach Laufzeit einfach kaputtgehen (wie es mir scheint, die zeigen dann "oberhalb von etwa 50% real" erstmal 20% "Zuschlag" mit an und oberhalb von 70% ist dann immer 99%), dann ist das wirklich schade. Ich vermute inzwischen fast, dass es BME280 auch als (schlechte) Fälschung gibt. 
Dann müssen wir uns mal nach eine hinreichend vertrauenswürdigen Bezugsquelle umsehen, die langzeitstabil ist. 
Denn "ich bestelle bei ebay-Händler in Deutschland" scheint nach den bisherigen Erfahrungen der Kollegen eher der Garant dafür zu sein, die "photosensiblen" zu bekommen, die zudem auch noch den Druck-Offset haben.

msebald, 2017-11-22:
Hallo und Danke für Deine Antwort. Soweit ich das beurteilen kann, haben die drei genannten Sensoren keine Berührungspunkte mit Leuchtstofflampen. Mein Sensor ist auf dem Dach, da kann maximal die Straßenbeleuchtung, die sich in ca 15 Metern Entfernung befindet, indirekt innen in der Röhre für etwas Helligkeit sorgen. Alle drei Sensoren sind erst seit diesem Jahr in Betrieb, ich schätze mal ein halbes Jahr. Wobei ganz zu Beginn waren die Werte gefühlt besser, aber auch zwischendurch mal. Aber 6 Monate reichen doch nicht aus, um den Luftfeuchtigkeitssensor so dermaßen altern zu lassen, oder? Bestellt habe ich in China, wie auf luftdaten.info vorgeschlagen. Habe, wenn es zum Ziel führt, aber kein Problem, einen vermutlich kleinen Betrag zu investieren und das Modul neu zu kaufen, gerne auch anderswo.

msebald, 2017-11-22:
PS: Die drei genannten Sensoren verwenden das DHT22 Modul, nicht BME280. Sorry, im Eingangspost und auch in der Antwort eben vergessen zu erwähnen. Das Thema Leuchtstoffröhren ist also damit eh vom Tisch.

Adorfer, 2017-11-22:
o.k. wenn's DHT22 ist: Das von Dir beschriebene Verhalten habe ich systematisch auch bei mehreren Sensoren gesehen. 
wie beschrieben: oberhalb von 50% gibt's ca. 20% "Zuschlag", also faktisch ein Springen von 50% auf 70%... und dann ziemlch schnell 99% obwohl der benachbarte BME380 meint "78%". 
Also ein sehr "eckiges" Verhalten beim DHT280. 

z.B heute real laut BME280 (was ich für einen plausiblen Verlauf halte)
[image]
vs. Output vom DHT022: 
[image]
(Die beiden Sensoren sind "rücken an Rücken" mit Kabelbinder zusammen im Rohr)

pathmapper, 2017-11-22:
Ich habe mit meinem DHT22 auch fast immer Luftfeuchtigkeitswerte von 99,9%.
Werde jetzt 'mal den von @Adorfer in https://github.com/opendata-stuttgart/sensors-software/issues/137#issuecomment-331731434 vorgeschlagenen  "BlueberryE BME280" ausprobieren. Gehe jetzt 'mal davon aus, dass die schon ordentliche BME280 verarbeiten werden (ist aber vielleicht auch nur Wunschdenken).

 
Luftfeuchtigkeit ist schon wichtig, da sie laut dieser [Untersuchung](http://www4.lubw.baden-wuerttemberg.de/servlet/is/268831/messungen_mit_dem_feinstaubsensor_sds011.pdf?command=downloadContent&filename=messungen_mit_dem_feinstaubsensor_sds011.pdf) einen Einfluss auf die Genauigkeit der SDS011 Messerwerte hat.
Insofern wäre eine verlässliche Messung der Luftfeuchte sehr sinnvoll. 

Generell wäre es gut mehr Informationen zu verlässlichen Bezugsquellen zu sammeln. Vielleicht gibt es auch 'mal eine Alternative zu "fast jeder lässt sich seine Komponenten einzeln aus China per Luftfracht kommen", gibt bestimmt ökologisch bessere Lösungen gerade für ein Projekt welches sich der Luftqualität widmet.

msebald, 2017-11-22:
@Adorfer Das sind spannende Informationen und Graphen. Danke dafür! Heißt für mich schlicht, dass der DHT22 für den Verwendungszweck nicht ansatzweise taugt, da das Problem ja weitverbreitet ist. V.a., wenn man diese Untersuchung, die @pathmapper (danke auch für Deine Informationen) hier aufführt. Auch, wenn das eigentlich an anderes Thema ist: Fließt die Luftfeuchtigkeit aktuell in die Ermittlung der Feinstaubwerte mit ein? Und beide Antworten wären (v.a. beim Einsatz von DHT22) fatal...

Ich bin nun bissle ratlos. Den von @Adorfer erwähnten BME280 Sensor kann ich mir schon kaufen, kein Ding. Gibt es aber eventuell noch sinnvollere Alternativen? Habe nach Installation und Flash der Firmware schon lange nicht mehr in die Configseite der Weboberfläche geschaut. Da hat sich ja einiges getan, man kann zig Sensoren anschließen, wobei ich nun nicht recherchiert habe, welcher für was ist. Zumindest beim HTU21D steht in Klammern dabei, dass er für Temperatur und Luftfeuchtigkeit ist.

@pathmapper, stimme Dir zu. Eine abrufbare Bezugsquellenliste mit Empfehlung wäre sinnvoll. Sich 6 separate Sendungen aus China zusenden zu lassen und das jeder, der so ein Ding baut, das ist sicherlich umwelchttechnisch nicht sehr nachhaltig. Und wenn's dann Schrott ist, umso weniger. Womit wir beim nächsten Thema sind, das mit meinem Issue nichts zu tun hat: Ich habe außer dem Github Issuetracker und direkten Mails an die Macher (leider ohne Antwort oder sehr zögerlich - was ich aber auch grundsätzlich verstehe) leider keine Möglichkeit gefunden, mit den Machern und der Community in Kontakt zu treten. Hier fehlt imho ein Forum oder eine Mailingliste. Oder hab ich das einfach noch nicht gefunden?

ricki-z, 2017-11-22:
@msebald: wir versuchen, uns auf wenige Kontaktmöglichkeiten zu beschränken, die aber möglichst alle nutzen können. Wenn jeder von uns ständig auf Dutzenden Kanälen schauen müsste, wäre die Kommunikation noch schlimmer ... An wen hast du denn EMails geschickt? Manche Mails bleiben im Spamfilter hängen und werden dann evtl. erst tage später gelesen.

Wir wollen die Anleitung (auch auf Grund der Probleme mit dem DHT) auf den BME280 umstellen. Wir hatten im Frühjahr noch die Hoffnung, das der BME680 zu einem vernünftigen Preis erhältlich ist. Dieser misst zusätzlich noch VOC (Volatile organic compounds), was für die Luftqualität interessant gewesen wäre. Der BME680 hat sich aber immer wieder verzögert, weshalb wir auch die Umstellung noch nicht gemacht haben.

Derzeit gehen die Luftfeuchtewerte des DHT nicht in die Feinstaubwerte ein. Angedacht ist, die Daten des Deutschen Wetterdienstes zu nutzen, solange wir nicht genau wissen, wie gut die Luftfeuchtewerte der einzelnen Sensortypen sind.

Mailinglisten: Tendieren dazu, das 90% der Mails für den jeweiligen Empfänger nicht von Interesse sind. Oder man muss für einzelne Themen Mailinglisten einrichten. Was dann auch wieder dazu führen kann, das Mails nicht von den richtigen Leuten gelesen werden, weil diese das Thema nicht abonniert haben ...
Forum: Wenn man 'Issues' nicht ganz so eng auslegt, kann man das als Forum nutzen ;-)

Sonst nutzen wir Slack für die Kommunikation. Das ist aber wahrscheinlich nur für Leute interessant, die des öfteren Fragen haben.

msebald, 2017-11-22:
@ricki-z Ich hatte Jan angeschrieben. Finde ich nun grundsätzlich ja nicht so schlimm, habe dann ja, als ich auf die Repos hier in Github aufmerksam gemacht wurde, hier einen Weg gefunden. Man muss aber sagen, dass es nicht ganz so einfach ist, mit Euch in Kontakt zu treten. ;-) Ne klare Call to Action auf der Webseite vermisse ich da. Und solch ein Projekt lebt ja von der Beteiligung.

Foren, Mailinglisten und was man sich sonst noch alles vorstellen kann, das waren nur spontane Vorschläge. Foren und Mailinglisten haben den Vorteil, dass die Informationen zugänglich vorliegen und es im Gegensatz zu Mailkontakt zu Euch nicht zu den immer gleichen Fragen kommt. Theoretisch. ;-) Foren ist Pull, Mailinglisten ist Push. Das kann man sehen, wie man will. Die einen wollen's so, die anderen so. Slack benutzen sicherlich nur wenige.

Github wäre noch am ehesten wie ein Forum, das schreibst Du ja schon. Allerdings lange nicht so komfortabel etc pp. Ein Link auf der Webseite mit einer kurzen Beschreibung wäre aber sicherlich sinnvoll, oder?

Und wegen meinem Issue hier: Gut, ich werde dann mal den hier empfohlenen BME280 anschaffen. Auch, wenn die Werte aktuell nicht zur Berechnung der Feinstaubdaten herangezogen werden, ich benötige diese eigentlich für meine Hausautomation - oder muss sie dort in meinen Berechnungen ignorieren. Luftfeuchtigkeit (erstmal) vom Wetterdienst zu holen denke ich ist sinnvoll. Das kann man umstellen sobald man weiß, eine valide Quelle für Luftfeuchtigkeit gefunden zu haben. BME680 und VOC, das klingt spannend! :-) Aber wäre wieder ein anderes Thema. ;-)

pathmapper, 2017-11-22:
> Ich habe außer dem Github Issuetracker und direkten Mails an die Macher (...) leider keine Möglichkeit gefunden, mit den Machern und der Community in Kontakt zu treten. 

Ja, finde es auch sehr mühsam sich mit Informationen zu versorgen, sich über den aktuellen Stand und die Entwicklung des Projektes zu informieren und sich aktiv einzubringen.
Informationsquellen die ich neben Github noch nutze:

- Twitter: https://twitter.com/codeforS

- Protokolle im [Meta-Wiki ](https://github.com/opendata-stuttgart/meta/wiki), z.B. [hier](https://github.com/opendata-stuttgart/meta/wiki/2017-11-14) oder [hier](https://github.com/opendata-stuttgart/meta/wiki/2017-10-24).

- Blog (leider nicht so oft neue Infos): http://luftdaten.info/news/

Da finden sich oft interessant Sachen, aber es bleiben auch viele weitergehende Fragen dazu unbeantwortet.

Mein Eindruck ist, dass es mit dem Projekt schon mächtig vorwärts geht und es auch eine sehr aktive Community im Stuttgarter Raum gibt (wo das Projekt ja auch seinen Ursprung hat).

Ich denke das Projekt ist in letzter Zeit ziemlich stark gewachsen und entsprechend viel haben die Macher auch derzeit um die Ohren.

> Hier fehlt imho ein Forum oder eine Mailingliste. 

Ich finde die Idee mit der Mailingliste einen Versuch wert, Github nutzten nicht alle Personen, aber eine Email kann jeder schreiben.

msebald, 2017-11-22:
@pathmapper Ich würde eine Mailingliste mit öffentlichem Archiv auch gut finden, ein Forum allerdings noch viel besser. Dieses bietet halt einfach mehr und auch modernere Möglichkeiten. Eine Webseite ist ja ähnlich universell wie eine Mail.
Aber letztendlich müssen das die Macher entscheiden, es gibt ja sicherlich auch Gründe, warum das bisher anders gehandhabt wurde.
Ich denke halt, dass mit so einer Kommunikationsplattform wieder gebundene Zeiten freigespielt werden und allgemein das Projekt vermutlich noch mehr Drive aufnehmen könnte.

Bezüglich des Sensors @Adorfer: Der BlueberryE BME280 für rund 10 € taugt im Gegensatz zu den BME280 für rund 3 € aus China wirklich so viel mehr, dass sich der Mehrpreis lohnt? Du hast ja schon die Hauptprobleme beschrieben. Wenn diese nur mit den günstigen Chinaversionen auftreten, ist der Mehrpreis gerechtfertigt.

Adorfer, 2017-11-23:
@msebald 
1) Wenn die Chinaquelle "3USD" für den BME280 sagt, dann schau sicherheitshalber genau hin, ob das BME280 evtl nur im Titel steht und untem im Text dann klein, dass aber nur ein BMP280 geliefert wird. Du wärst zumindest nicht der Erste, der sich hinterher "wundert". 

2) Den BlueberryE habe ich herausgesucht, weil behauptet wurd, es gäbe den BME "nur zum Löten und daher sei er ungeeignet". Es scheint mir aber eine Quelle zu sein, bei der auch Hardcore-Breadborder glücklich werden. Ich selbst habe dort noch nie gekauft.

3) Mailingslisten landen bei mir aufgrund der chronischen Vollquoterei fast immer im Spam. (Nachrichten mit >50% Quoteanteil können nicht sinnvoll sein.) Wenn ihr ein Forum wollt, nehmt ein Discoure, auch wenn diejenigen, die das betreiben sollen, damit Schmerzen haben werden. Oder fragt bei einer befreundeten Organisation, ob ihr da ein Unterforum bekommt.  (Wenn wir das weiter diskutieren wollen, bitte ein neues Issue. Das soll mein letztes zu dem Thema hier sein.)

dokape, 2017-11-23:
Oben beschriebenes Verhalten mit oft 99.9% Luftfeuchte hatte ich auch nach dem Öffnen und wieder Zusammenbau meines vorher ordentlich messenden Sensors. 
Ein paar Tage später habe ich den Kabelverhau durch eine kleine Platine ersetzt, auf dem DHT22, BME280 und nodeMCU aufgesteckt sind. 
Seitdem liefern DHT und BME nahezu gleiche Werte. Das Sprunghafte Verhalten des DHT22 ist nicht mehr erkennbar. 
Da beim DHT22 die Anschlüsse sehr dünn sind, kann ich mir einen schlechten Kontakt durchaus vorstellen, der für schlechten Werte möglich ist.

msebald, 2017-11-23:
@Adorfer Thank you for your information. I will order a BME280 and will let people here know how everything worked out after installing it and running it for a while. For the discussion platform topic I opened a issue [here](https://github.com/opendata-stuttgart/luftdaten.info/issues/5).

@dokape Klingt nach einigem Gefummel und viel Arbeit. Gibt es da ein paar mehr Infos, Fotos oder mehr?

dokape, 2017-11-24:
Kleine Platine. Ist nicht wild. 
Gefummels war, die Stiftleiste vom USB Adapter SDS abzulöten und auf die neue Platine zu löten. 
Bilder hab ich leider keine gemacht.

dokape, 2017-11-25:
perhaps your DHT22 is broken? 

It does not look like a software-problem. 
My actual values looks fine (it's November Rain outside): 

[image]

msebald, 2017-11-26:
@dokape Sure, it might. Yours looks just fine. As I have bad values and as I noticed that 2/3 of all devices have bad values (the other 1/3 might use the BME280) I opened this issue to find out of it is software or hardware related. If second it looks like there is a general problem with DHT22. So changing the hardware might work, but there seem to be better solutions to the problem here.

dokape, 2017-11-26:
As my values also changed from normal work to 99.9% after a maintenance and went back to normal operation after another maintenance, it could be possible the location in the airrohr or some bad connection - oh well, then temp should be also bad - could cause that values. 
The BME 280 is at about same price. But you have to solder the 4 pins to the board.

sandboxgangster, 2017-11-28:
@dokape @msebald  Soweit ich weiß kommt der DTH22 nicht mit dauerhaften hohen werten klar, meiner ist auch nach 2 Monaten ausgefallen (Hamburger Nebelwetter) hab mittlerweile auch ein BME280 (Chinaversion) verbaut mal sehn ob der besser damit klar kommt.

Adorfer, 2017-11-29:
it is recommended to use a BME280 with "onbard 3.3V-regulator / for feeding with 5V". Those can cope better with noise/ripple from the power supply or when operating on longer/unschielded cables. (i do not say, "the 3.3V ones do not work", but "5V supply ones" work seem more reliable in some environments.

svenkubiak, 2017-12-11:
Currently having the same issue with DTH22 and mostly constant 99% value.

[image]

@msebald 
Any news on your experience with the BME280?

dicer, 2017-12-11:
Ich habe letztes Jahr von DHT22 auf BME280 umgestellt, weil ich genau das gleiche Problem hatte (allerdings im Ausland mit hoher genereller Luftfeuchtigkeit). Seit dem habe ich keine Probleme mehr.

Ich verwende einfach die billigsten Breakout-Boards von Aliexpress und die funktionieren auch draussen bestens. Hab 3 draussen und 4 innen im Einsatz.

msebald, 2017-12-13:
@svenkubiak I just installed the BME280 today. It shows realistic humidity values. I compaired these with other devices I have outside. DHT22 jumped to 99% again right after starting the device.

I will post some graphs as soon as I have it running for some days. I did not remove the DHT22 sensor to see the difference.

msebald, 2017-12-14:
The sensor has no contact with the housing. I attached it to one of the boards. Will take a close look to the value though. It also rained a lot the last night and the values look great.
[image]

dokape, 2017-12-15:
My suggestion: try another DHT22.

msebald, 2017-12-15:
Could help, but might not. As many people notice problems with the DHT22 sensor. Of you browse through the Feinstaubsensor devices on the map many many (roughly about 50%) have these 99% issues.

So I try the BME280 and gather experience on that sensor. Lets see how values are with this one in 2-6 months.

Adorfer, 2017-12-15:
Just be warned, that the BME280 is - in my oppinion - just better / more reliable than the DHT22, but still far away from beeing perfect. So you may run into the "99%"-Scenario too, just far less often.

dokape, 2017-12-16:
I had the 99% too.
But at the Moment, DHT and BME280 are running with nearly same values.  

Imho the feinstaubmap and API should be channged.
At the moment the values of DHT are preferenced. 
Imho it the Sensor should deliver something like "temp=23; Sensor=DHT22";" temp=23.2;Sensor=BME280"
So if one Sensor is not available, the value o  the other sensor should be taken to display.

svenkubiak, 2017-12-25:
I recently installed a new DHT22. Numbers are still high, but it does not pin at 100%.

[image]

goggel, 2018-02-05:
Has anybody long term experience with the SI7021? It has a heating element to reduced for accumulated high humidity.
Funnily enough you can also bake the sensor at 125°C for 12 hours according to the data sheet.

coelner, 2018-02-20:
@goggel  Currently not. But the SI7021 is available with a PFTE cover, which protects the humidity sensor from environmental inputs like dust or fog. ( some versions of the SHT3X have this feature also)
you could find a detailed comparison of sensors here: http://www.kandrsmith.org/RJS/Misc/Hygrometers/calib_many.html
Another sensor which is prepared for outdoor use: https://www.adafruit.com/product/1293
I can't find any heating element in the bme280 or any information about behaviour in foggy conditions (except under condensation it could be stored powerless...). Maybe an added pfte foil improve the behaviour of all sensors.

dokape, 2018-03-14:
Is there still an issue with the DHT22? It seems running now nearly 2,5 Month at @svenkubiak 
@msebald can this issue be closed? 
@goggel this is an issue with DHT22, please follow up discussion for SI7021 at a specific thread for that. thanks.

svenkubiak, 2018-03-14:
Can be closed from my side.

JsBergbau, 2019-02-18:
Zum DHT22 hatte ich einen Test gelesen, der ihm relativ genaue Werte bescheinigte. Ich halte es für möglich, dass die Sensoren mit der Zeit "an der Luft" derart altern , dass die angezeigten Luftfeuchtigkeitswerte immer weiter steigen. 

Habe gerade einen BME280 mit Spannungswandler durch einen BME280 ohne Spannungswandler verbunden. Kurz vor 19:30 sieht man Spitzen in der Temperatur bzw. einen senkrechten Abfall in der Luftfeuchtigkeit. Das war der Tausch. Danach laufen die Luftfeuchtigkeitskurven des DHT22 und BME280 deutlich auseinander.

[image]

Der DHT22 ist seit Beginn in der Messtation und das ist seit Juli 2017.

Folgende Daten sind am PC mit einem Raspberry-PI aufgenommen worden:

Daten des BME280, der seit ca. April 2018, also weniger als 1 Jahr seinen Dienst tat. 
```
pi@raspberrypi:~ $ ./bme280.py
Chip ID     : 96
Version     : 0
Temperature :  21.95 C
Pressure :  987.561269783 hPa
Humidity :  57.0930918761 %
```

Daten des baugleichen BME280, der damals mitbestellt wurde und seitdem ungenutzt herumgelegen hat.
```
pi@raspberrypi:~ $ ./bme280.py
Chip ID     : 96
Version     : 0
Temperature :  22.43 C
Pressure :  987.728540843 hPa
Humidity :  38.3093445447 %
```
Ein nebenstehendes TFA https://www.idealo.de/preisvergleich/OffersOfProduct/588885_-thermo-hygrometer-versch-farben-tfa-dostmann.html zeigt 22,4 °C und 41 % Luftfeuchtigkeit. 
Der niedrige Wert um 40 % entspricht wesentlich mehr der Realität.

```
pi@raspberrypi:~ $ ./bme280.py
Chip ID     : 96
Version     : 0
Temperature :  21.41 C
Pressure :  986.290990444 hPa
Humidity :  47.7158411907 %
```
Dieser BME280 wurde ziemlich heiß als I2C am Raspberry-PI noch nicht aktiviert war und er dadurch mit GPIO Spannung verbunden war. Entweder hat ihm das so zugesetzt, dass er zu hohe Werte anzeigt, oder die Serienstreuung ist sehr hoch oder ich habe Fake-China-Ware erhalten. Ein weiterer BME280 aus dieser Bestellung zeigt mit dem TFA vergleichbare Werte an.

ricki-z, 2019-11-13:
@msebald Is this issue solved so we can close it?

dirkmueller, 2019-11-27:
We should capture the great amount of useful information in this issue in a more permanent place ..

msebald, 2019-12-14:
> @msebald Is this issue solved so we can close it?

Well, I solved it by adding a BME280 sensor and using the data from this sensor. I still have the DHT22 sensor up and running and it almost always shows 99.9%.

dirkmueller, 2020-01-04:
I've added this information to the Readme.md file. I might consider adding a broken-DHT22 detection in the future and then stop reporting its values (better not flood opensensemap and all other places with useless bogus information)

commit 1dad1a7c (2020-01-05): Add information from DHT22 debugging (Fixes #170)
closed 2020-01-04 by dirkmueller