opendata-stuttgart/sensors-software #171: Recommended Sensor

svenkubiak, 2017-11-27:
Hi,

I have seen that in the latest software release (NRZ-2017-099) a lot of sensor have been added.

SDS011
PMS1003, PMS5003, PMS6003, PMS7003
PMS3003
DHT22 
HTU21D
PPD42NS
BMP180
BMP280
BME280
DS18B20
GPS (NEO 6M)

Is there a list of functionalities and advantages on all of those sensors? Is there a recommended list of sensor to use for air quality, temperature, air pressure and humidity?

Thank you and best regards,
Sven

Adorfer, 2017-11-27:
we can start with the deprecated ones like PMS1003, PMS3003 due to dust accumulation issues. Pms5003 for having an easy soldering cable, but beeing bulky. 
PPD42NS for beeing outdated,
BMP280 for missing values compared to BME280.

svenkubiak, 2017-11-27:
@Adorfer 
Thanks for the heads up.

I currently have a SDS011 and a DHT22 in use and would like to have air pressure as well. Based on your infos I am thinking about an BME280 to have one sensor for temp, pressure and humidity. And as I am working on the station anyway, would it make sense to also update the air quality sensor to a PMS7003?

dokape, 2017-11-27:
Standard Setup is actual:
SDS011 and DHT22 (easy to wire, no soldering, just put cables together. Easy for nearly everyone)
Works fine.

For more advanced "makers" you can add the BME280. You have to solder the 4 pins. 
Then use cables to put everything together. 
The BME provides humidity in a little bit better quality as the DHT, provides temp and additional provides
air pressure. But: It provides the air pressure AS IT IS at your place. 
If you are are higher then the sea, you get lower values. eg. At about hight of 400m, you are about 50hPa lower than at hight of 0m. (the difference depends on humidity and temp!).
So using the air pressure will not give you a direct comparable value for your weather forecast! 

The PMS-Sensors are cheaper than the SDS011, they are supported by the software to have another
sensor as the SDS011 would not be available at your place. They are smaller, if you are limited in space.

DS18B20 is a very often used sensor. Often this is provided with some aduino boards. 

I have DHT22 AND BME280 on my actual board. Runs fine. But do not overweight the air pressure ;-)
With the BME280 you have a good choice. 

PMS7003: If you buy one, have a look that a cable is provided by the seller.
I bought mine without a cable. The pins are in 1.27mm distance, not the 2.54mm distance.
IMHO for getting comparable values, it should be good, when most nodes have the same (SDS011) dust Sensor.

svenkubiak, 2017-11-27:
@dokape 
Thanks a lot!

> IMHO for getting comparable values, it should be good, when most nodes have the same (SDS011) 
dust Sensor.

Makes sense. Totally agree.

Which one is the correct BME280?
https://www.adafruit.com/product/2652
https://www.reichelt.de/Kombosensoren/BME-280/3/index.html?ACTION=3&GROUPID=6845&ARTICLE=159825

ricki-z, 2017-11-27:
From the Luftdaten.info team:

Recommended PM sensor is the SDS011. All other PM sensors were included for comparison tests or for users that will use our firmware for own projects. I.e. a belgian group will use the Honeywell sensor but wouldn't write a new firmware.

Recommended temperature/humidity sensor is the DHT22 or the BME280 if possible. The BME280 has a higher price and/or you need to solder the connector pins/cables.

ToronMega, 2017-12-03:
Hello,
great project.

I think it would be great to integrate this sensor (MQ-2 -> Gas, Smoke).

Because my idea behind it is to recognize what kind of "pollutants" are in the air.
For example, if an neighbor has just started his carmine again, or something similar.

My house has a KWL, and I would like to air the house only when the air is at its best...

Regards

Adorfer, 2017-12-03:
there is to contraints to MQ2 and other organic sensors: 

- limited lifetime
- high power consumption due to required heating

ToronMega, 2017-12-03:
I understand that, but lifetime of MQ2 or electricity bill are a minor problem, when your neighbour  is  burning plywood in his fireplace. 

And the great steam is sucked into the house by the KWL or an open window.

Regards

ricki-z, 2017-12-15:
@ToronMega 
Note from 'Seeed studio' website:
"The sensor value only reflects the approximated trend of gas concentration in a permissible error range, it DOES NOT represent the exact gas concentration. The detection of certain components in the air usually requires a more precise and costly instrument, which cannot be done with a single gas sensor. If your project is aimed at obtaining the gas concentration at a very precise level, then we don’t recommend this gas sensor."
We will add support for the Bosch sensor BME680. This sensor will measure VOC (Volatile organic compounds) and will give an Air Quality index for this. Most of the gases detected by the MQ2 can be detected by the BME680 also. And the BME680 can measure temperature, humidity and pressure also.
Would this be okay for you?

ToronMega, 2017-12-17:
Sounds good :-) I am curious. I look forward to the implementation.

Thx :-)

coelner, 2018-02-20:
keep in mind that the bme680 is for INDOOR quality, not outdoor air quality. usually official stations measure PM10 (not PM2.5), NO2, O3, CO and SO2 (https://en.wikipedia.org/wiki/Air_quality_index#CAQI). As seen here : http://www.libelium.com/development/waspmote/documentation/gases-board-technical-guide/ there are a bunch of sensors which could be added to determine a general air quality, especially in cities. The MQ-X winsen sensors are tricky to use because old fashioned analog sensors. Therefore I would use the sensors from sgx, at least the do not need a heating cycle or relatively high heating voltage.
- NO2: SGX MiCs 2714 (https://sgx.cdistore.com/ProductDetail/MICS2714-SGX-Sensortech-Limited/333416/)
- O3: SGX MiCs 2614 ( n/a), Winsen MQ-131 (needs 6V! to heat)
- NO2/CO: SGX MICS-4514 (https://sgx.cdistore.com/ProductDetail/MICS4514-SGX-Sensortech-Limited/333417/)

Usually you need to replace those sensors after 6 months, humidity and harsh environmental load destroy them.
O3 and NO2 are very similar, even if we determine only a drift in these things, we need two sensors to correlate the measurements. 
SO2 sensors cost about 100€/USD

constant5, 2018-07-31:
I would like to integrate a CO monitor into a couple of these PM monitors for a short research project.  Is there some script to add another analog output voltage to the NodeMCU?

pluisi, 2018-07-31:
Nodemcu has only one analog input.
You could buy additional boards and/or ICs called ADCs that can read analog
signals into digital ports.

A totally partial list follows:
- ADS1015 12 bit
- ADS1115 16 bit
- MCP3208 12 bit
- MCP3304 13 bit
(Taken from the following link
https://arduino.stackexchange.com/questions/25968/using-an-external-adc-with-arduino
)

Good luck
Pier

constant5, 2018-08-01:
Thanks Pier. Do you think the ADC is better than using a multiplexer?

pluisi, 2018-08-01:
A multiplexer can have problems related to  signal conditioning if readings
are too fast, while the addADCs strategy has the only limitation of the
sampling frequency.
Just think of your application and how fast are giving a read. Make your
calculations on costs. Decide what's better.
Have fun
Pierluigi

step21, 2019-07-23:
Hi,

hope it is okay to still post here - what do people think about this shield advertised for raspberry pi etc. - https://shop.pimoroni.de/products/enviro-plus in conjunction with the pms5003 https://shop.pimoroni.de/products/pms5003-particulate-matter-sensor-with-cable ?
Not the cheapest but it has a lot of sensors. I read above that the pms5003 is not recommended/deprecated due to being bulky, but to me that seems like minor concern/non-issue. I am more concerned about how to hang the sensor, protect it from rain and so on. Especially as the pm sensors or at least the pms5003 have a fan + airflow, so this would somehow have to be protected as I guess it is not ideal if it rains into the fan or similar...
How is this handled normally?

MagTun, 2019-09-16:
> We will add support for the Bosch sensor BME680.

@ricki-z, are there any news about this?

dirkmueller, 2020-01-04:
see #411  for tracking the BME680 support. I've ordered one and started adding it, but the added code size was throwing me off..  will look at it again shortly.