opendata-stuttgart/sensors-software #173: Firmware compatible with other ESP8266 based boards?

fredg02, 2017-12-04:
Hi,

I did no find any info about this in the wiki or in the GitHub issues.
Has anyone tried to run the firmware on other ESP8266 based boards?
E.g.
* [Adafruit Feather HUZZAH ESP8266](https://www.adafruit.com/product/2821)
* [SparkFun ESP8266 Thing](https://www.sparkfun.com/products/13231)

Are any major changes in the firmware to be expected to make this happen?

I'm aware that those boards are not as cheap as the NodeMCU boards, but they have some nice features like on-board LiPo connectors and chargers. The HUZZAH is also extensible with "FeatherWing" boards (e.g. RTC + SD-Card slot for logging data).

Regards,

Fred

Adorfer, 2017-12-04:
I used "WemosD1 Mini" and see no reason why others shouldn't.
Only beware of esp modules with small flash sizes, there are some around.

ricki-z, 2017-12-04:
The firmware should work with most ESP8266 boards. Flash size should be 2MByte minimum. The pin to gpio mapping may vary so you have to check how the components are connected.

ricki-z, 2017-12-04:
Having connectors for components like SD-Card or RTC clock doesn't mean that they are working all together. The GPIOs of the ESP8266 are very limited and most of the free GPIOs are used by other components in our firmware. ALso you have to extend the firmware by yourself if you plan to use any new components. This may lead to an incompatible firmware version.

Adorfer, 2017-12-04:
a migration to ESP32 could be a way out of the problem, since ESP32 has significantly more IO, so the risk of collision of functions on "well equipped boards" is far lower. In plus the flash is more spacious and for future "solar" models, it has lots of more sophisticated sleep modes. 

In other words: if somebody of you considers of "forking firmware in order to make more/different IO/boards": Please evaluate doing the step in direction ESP32, since it eliminates some other restrictions as well.

fredg02, 2017-12-05:
@ricki-z Thanks for the feedback. I just wanted to know if it's possible in general first. Things like SD-card or RTC would only be nice-to-have. The LiPo charger and voltage measurements are the more interesting part.
@Adorfer I will look into ESP32 boards as well. Since I don't have much experience with microcontrollers, porting the code is probably a bit over my head though.

Maybe that info ("yes, in general it should run on other ESP8266 boards") should be added to the readme and/or the wiki?

DwaineGarden, 2018-03-24:
I have the Adafruit Feather HUZZAH ESP8266, Adafruit Plantower PMS5003 and Adafruit DHT22 sensors.  I see that the firmware code you have to edit ext_def.h for the pinout.  What other changes are needed to get the firmware working with the Plantower PMS5003?

I have both Plantower PMS5003 and the Adafruit DHT22 working fine with a basic code.

[image]

ricki-z, 2018-03-25:
For your own firmware you should deactivate the auto updates (by setting AUTO_UPDATE to 0), set the PMS32_READ to 1 and the SDS_READ to 0.

Adorfer, 2018-03-26:
I do not get the question about the adafruit config.
at least for PMS5003/PMS1003 it is working since (at least version 099) "out of the box" on esp-boards with (minimum...) 4MB flash. 
No need for any local reconfiguration.

DwaineGarden, 2018-03-27:
The pins are different

Adorfer, 2018-04-11:
the pins ware by default not soldered in, but you have it yourself, right?

DwaineGarden, 2018-04-23:
I'm happy with the project and the LED notification is working great!   The Plantower PMS5003 been running without a problem.
[image]

Adorfer, 2018-04-24:
what kind of housing is this?

DwaineGarden, 2018-04-24:
4 inch black PVC pipe end cap
4 inch black PVC drain end cap
4 inch white PVC section
One Lawn LED Light

Bought it at Home depot.

Adorfer, 2018-04-24:
o.k. it's just ping... i thought, this (including the lawn light) was one assembley "ready made" which you just siliconed in order to waterproof it.

DwaineGarden, 2018-04-25:
It takes about 10 seconds to glue the pieces together.  It turned out to be a very nice enclosure.

DwaineGarden, 2018-06-02:
Final picture of the sensor attached to the house
[image]

jordanc365, 2019-06-24:
Hey @fredg02  did you ever end up figuring out how to write to an SD card?

Adorfer, 2019-06-24:
prices for ESP8266 boards with intetraged SD-card slot are (in my eyes) ridicously high compared to the use of a cheap external slot (for arduino etc). Did you find any that is below 15€?

jordanc365, 2019-06-24:
Yeah the intergrated ones are crazy...
I've tried a few that work but I'm not good enough at coding to intergrate them into this project.
As a workaround it'd be pretty easy change the serial csv writer to write to a sd card.
https://www.ebay.com.au/itm/Micro-SD-Wemos-D1-Mini-Data-Logger-Shield-RTC-DS1307-Clock-For-Arduino-Raspberry/292929364924?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

[https://www.ebay.com.au/itm/Data-Logging-SD-Card-Socket-Shield-with-RTC-Real-Time-Clock-for-Arduino-WS/182497426363?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649](url)

fredg02, 2019-06-27:
> Hey @fredg02 did you ever end up figuring out how to write to an SD card?
No. As mentioned before, I was more interested in the LiPo charger and voltage measurement.

dirkmueller, 2019-11-27:
Don't see open questions or issues here, closing.

closed 2019-11-27 by dirkmueller