opendata-stuttgart/sensors-software #175: Feature: Support Syslog-server

dokape, 2017-12-17:
The airrohr FW supports debugging. Well, just debugging at the serial/USB-Port.
Most installations, the connection to a serial log is not very easy to manage. 
The airrohr is installed outside.

What about a support of sending the debug infos to a syslog server? 
e.g. running on a raspberry pi in the internal network? 

What do you think about this?

ricki-z, 2017-12-18:
We can try to implement this. But syslog would only work after connecting to wifi. And we have to look for memory consumption. Compiled with Release candidate 2 of the esp8266 libs we need nearly 48 percent of available flash memory. The limit is 50 percent if we use auto update.

dokape, 2017-12-18:
I See a benefit in logging failures by long usage like the bug #172 or log memory/stack usage oder long term. I see no need in implementation at once. Just a Diskussion, if needed vor not.

sandboxgangster, 2018-01-25:
Why do not you do two ways update, when the update first a smaller image which makes the flash free, and which then loads the actual larger update?

ricki-z, 2018-01-26:
You could try to build a smaller file, but event an empty script needs 21% of the available flash space. We need to read the config (ssid and password) so SPIFFFS support is added. We need to connect to the wifi netword, so the complete wifi stack is added. And we need the update function, so also this will be added. We won't get so much more 'available' flash memory. 
With a two step update we have a new point of failure .

sandboxgangster, 2018-02-13:
Ok, that's understandable, I know that from the Tasmota FW, which does not use SPIFFFS and writes directly into the flash at fixed addresses, which will not override synonymous with update, I know myself there but not enough, but I found the one good idea to free memory.

Tasmota memory layout https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips

dokape, 2018-03-14:
Due to lack of performance I closed the issue.

closed 2018-03-14 by dokape