opendata-stuttgart/sensors-software #177: altes binary in beta-pfad notwendig?

dokape, 2017-12-30:
Im Beta-Fork 

https://github.com/opendata-stuttgart/sensors-software/tree/beta/airrohr-firmware

befindet sich noch ein 3,5MB großes altes binary einer alten Firmware:

ppd42ns-wificonfig-ppd-sds-dht.spiffs.bin

wird das noch benötigt?
Das wird bei jedem download als zip mitgeschleppt. 
Wenn man das nicht benötigt, könnte man es auch entfernen.

ricki-z, 2017-12-30:
Das ist das leere SPIFFS-Dateisystem (.spiffs.bin). Wenn man die Konfig hart überschreiben will, wird das benötigt. Ich würde es gern hier lassen, weil ich nicht weiß, ob und wie die Seite luftdaten.info mal umgebaut wird. Und wir haben von einigen Stellen auf diese Datei verlinkt.

dokape, 2017-12-30:
Ok. Das ist wichtig. 
Der Dateiname hat mich zu der Frage veranlasst. Wenn verlinkt, dann kann man den Dateinamen natürlich auch nicht ändern. 

Evtl. in der Doku/readme entsprechend vermerken? 
Das ist das, wenn man sich z.b. über die BasicAuth ausgesperrt hat man über die Arduino IDE das Dateisystem neu beschreiben/erzeugen muss? Das dauert aber doch länger, als nur das Image zu schreiben?

ricki-z, 2017-12-30:
Das Image, welches normalerweise geflash wird, überschreibt nur die Software, nicht die Konfiguration. Diese liegt im Bereich oberhalb des ersten MB. Flasht man also nur die Software, bleibt die Konfiguration bestehen. Sonst würde ja auch das Auto-Update nicht funktionieren.

ricki-z, 2017-12-30:
Zweck der Dateien steht jetzt in der Readme. Kann der Issue dann geschlossen werden?

dokape, 2018-01-02:
Es gibt hier eine Anleitung zum Löschen der Konfiguration im WIki:

https://github.com/opendata-stuttgart/meta/wiki/Konfiguration-der-Sensoren#l%C3%B6schen-der-konfiguration

führt man das durch, wird ein neues leeres .bin erzeugt (nach log der arduino)
Diese spricht aber nicht vom diesem binary: ppd42ns-wificonfig-ppd-sds-dht.spiffs.bin

Meine Frage/Schlussfolgerung: Gibt es mehrere Möglichkeiten, die Konfiguration zu überschreiben? 
Also im zweiten Falle das ppd42ns-wificonfig-ppd-sds-dht.spiffs.bin binary flashen wie eine normale Firmware und damit die Konfig überschreiben?

ricki-z, 2018-01-02:
Das steht im Wiki auf der Seite  "Firmware einspielen" als Variante 2 "Löschen der Konfiguration"
Unter Konfiguration sind nur die Klickvarianten aufgeführt. Da könnte man das nochmal aufführen, aber mit der Kommandozeile sind wohl einfach ziemlich viele Leute überfordert.

ricki-z, 2018-01-02:
Und ja, es gibt mehrere Varianten, die Konfiguration zu löschen.
1. Via integriertem Webserver: Punkt 'Konfiguration löschen' auf der Startseite. Geht aber nur, solange der Sensor erreichbar ist
2. Installieren des Plugins  "ESP8266 Sketch data upload" in der Arduino IDE. Dann kann ein leeres Dateisystem erzeugt und geflasht werden.
3. Direktes Flashen von ppd42ns-wificonfig-ppd-sds-dht.spiffs.bin , das ist das gleiche wie Punkt 2 ohne Installation des Plugins und mit Flashen über die Kommandozeile. Das leere Dateisystem ist ja immer das gleiche, deshalb muss sich ja nicht jeder das Plugin installieren.

closed 2018-03-14 by dokape