opendata-stuttgart/sensors-software #179: Use web page (like config.luftdaten.info) to configure the esp8266

dadosch, 2018-01-04:
I'm not sure if it is possible, but I explain what I mean:
You externalize the whole full-blown webpage (with styles+logos) to a host reachable from the internet (for example config.luftdaten.info).

On that site, you could enter the IP address of the esp8266. It then would send from the user's browser the config to a much simpler *local* interface on the esp8266.
(You still need to be on the same network to configure the esp)

**This would save quite a bit of memory.**

Only the wifi credentials must be entered directly in the esp8266. (Or use a HTML5-Offline-App)

[Cross-Origin Resource Sharing](https://de.wikipedia.org/wiki/Cross-Origin_Resource_Sharing) may be needed

Maybe I explained a bit weird what I mean, so here is a drawing:

[image]

ricki-z, 2018-01-04:
On a WPA2-enabled wifi the ESP8266 won't have an IP address until configured as it don't know the needed credentials.
So we have to implement a web page for configuration anyway. The pages, styles and logos (could be removed or made smaller) don't need so much more memory. With your solution we have to 'develop' just another piece of software/webpage that needs to be maintained.

dadosch, 2018-01-04:
Yeah, it was just an idea to get some space on the esp8266...

closed 2018-01-04 by dadosch