opendata-stuttgart/sensors-software #182: idea: further sensor for radio activity

SomeStrangeName, 2018-02-08:
I am asking me if the sensor could become a more generic sensor box for environmental data. Dust and temperature is already a very good thing but could also be interesting to add some radioactivity sensor. 

Looks like there is someone in Stuttgart building some sensor in that direction:
* http://www.opengeiger.de/elektronik.html

Maybe this sensor RD2007 http://skory.gylcomp.hu/alkatresz/RD2007_X-Ray_sensor.pdf could be a good starting point?

Nordrunner01, 2018-02-10:
Moin.
Auch hier können wir...... :-=
Guggst du : https://apollo.open-resource.org/lab:pigi
Und Bezugsquellen für die PiGi-Platine findest du hier : https://forum.iot-usergroup.de/t/bestellmoeglichkeit-pigi-platine-bestueckt/84

Da der PiGi am Ausgang einen RasPikompatiblen Impuls liefert, welcher an seine negativen Flanke gezählt wird, sollte eine Auswertung auch mit einer anderen MCU umsetzbar sein.
Außer die Geiger-Müller Zählerröhrchen, die in unserem Forum verlinkt sind, kann man auch ein SBM20 verwenden. Meine habe ich über die Bucht aus der Ukraine importiert. Kostenpunkt war meine ich 25€ für 2 Stck. Bin mir da aber nicht mehr sicher. Auf alle fälle musste ich zum Zoll und die Teile abholen, weil der Preis über der Freigrenze lag.
Witzig die Fragen was ich damit vorhabe!!!! :-D

Gruß

ricki-z, 2019-03-13:
closed because double of #212

closed 2019-03-13 by ricki-z