opendata-stuttgart/sensors-software #184: Troubleshooting a station that doesn't yield data any more

marians, 2018-02-09:
So I have this station 

    ID: 12174814
    https://www.madavi.de/sensor/graph.php?sensor=esp8266-12174814-sds011
    Particle sensor ID 6316 in database
    Temperature/humidity: 6317

that hasn't been sending any data for ages. At least it doesn't show up in the [list of active sensors](https://www.madavi.de/sensor/active_sensors.php), it has no data showing in the graphs and I also got an email ages ago that it's offline.

I would like to find out what's going on, but have no idea where to start.

The device is currently attached to my Computer's USB. When being connected to the cable, it automatically turns on for a few seconds, then turns off (particle sensor fan and LEDs go off). After ~ 2 minutes 40 seconds this repeats.

For some reason, I don't see the USB serial port in Arduino's `Tools` > `Port` menu. So I cannot select it to start the Serial Monitor (`Das Board an /dev/cu.wchusbserialfa130 ist nicht verfügbar`). The compute and USB serial driver is the same combination I used to flash the sensor device a couple of months ago. But I've had bad luck with some USB cables. Maybe it's the same again this time. Maybe I should switch the USB serial driver. Glad for any hints which one should work (Mac is on OS 10.12.6 Sierra still).

Is there any other way beyond the Arduino serial monitor to find out what the station does?

I would have some LCD display panel around somewhere. I'd have to find it in order to check what type it is. Would that be an option?

The station uses my unencrypted `Freifunk` wifi. Can I intercept the wifi traffic?

Here is some information that Arduino spits out:

Arduino: 1.6.12 (Mac OS X)
Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 9600, 4M (3M SPIFFS)"

On the board it says NoceMCU v3.

rexfue, 2018-02-10:
Hallo Marian,

gestern von 19:58 bis 21:16 hat dein Sensor gesendet, d.h. prinzipiell arbeitet der (siehe Grafik auf feinstaub.rexfue.de/6316 <http://feinstaub.rexfue.de/6316> )
Kann es sein, dass der Freifunk unzuverlässig ist?

 Zur Info: dass der SDS011 den Lüfter alle 2.5min anmacht ist so gewünscht: es wird nur alle 2.5min gemessen und dann wieder
Pause gemacht.

Beste Grüße
Reinhard

ricki-z, 2018-02-10:
Bitte auch schauen, ob die Madavi API in der Konfiguration aktiviert ist. Ohne dies werden auf Madavi.de auch keine Grafiken angezeigt.
Die Konfiguration sollte über die IP-Adresse mit einem Browsererreichbar sein. Die IP-Adresse findet man auf dem Heim-Router meist unter "Verbundene Geräte", der Name des Sensors beginnt dort mit 'ESP_'
Gruss Rajko

marians, 2018-02-10:
Danke @rexfue! Dass sind zwei wertvolle Neuigkeiten. Die, dass der Sensor offensichtlich funktioniert. Und dass es da noch eine weitere Anwendung zur Darstellung der Daten gibt, die ich bisher nicht kannte.

Die anderen Fragen bleiben für mich interessant und unbeantwortet. Weil das eigentliche Problem aber offensichtlich nicht mehr besteht, schließe ich das Issue.

closed 2018-02-10 by marians