opendata-stuttgart/sensors-software #186: Project: Designing a PCB

Naesstrom, 2018-02-23:
I'm designing a PCB to be screwed on to the SDS011 sensor with a Wemos D1 mini on the backside and connectors for the sensor.
I've also added connectors for some I2C and a DHT so far but are there any more connections I should add before ordering them?
I've read a thread here with suggestion to add adressable RGB leds and that would be awsome but then I would want to know what pin might that end up on etc?
[image]

Oh and for some encouragement for the LED, I rendered my case for it with a green LED at the top...
[image]

Hammarstrom, 2018-02-23:
Nice, have you tested a ws2812b as an option?

Naesstrom, 2018-02-23:
Nope, I'm not brilliant enough to understand how to change the code so that the values correspont to colors... I just like to tinker and invent things! 😄

ricki-z, 2018-02-24:
The SDS011 RX and TX lines are crossed (right of I2C connectors). So this won't work.
And th wifi antenna should be 'outside' the area of the SDS011. This will result in a better connectivity.
For the LEDs wo don't have any free IO pins. so we could only use LED stripes with an I2C interface.
For the LCDs an I2C connector with 5V on VCC would be nice.

Naesstrom, 2018-02-24:
Oh thanks for the heads up there @ricki-z, about the antenna, I thought it was enoough that it was pointed away from the sensor but I'll look at it again.

Is there a list of all the IO pins somewhere, I didn't think there were more then for the SDS, DHT and i2C in use?

ricki-z, 2018-02-24:
According to the Wemos Wiki The D1 mini has the folling pins:

Pin | Function | ESP-8266 Pin
-- | -- | --
TX | TXD | TXD
RX | RXD | RXD
A0 | Analog input, max 3.3V input | A0
D0 | IO | GPIO16
D1 | IO, SCL | GPIO5
D2 | IO, SDA | GPIO4
D3 | IO, 10k Pull-up | GPIO0
D4 | IO, 10k Pull-up, BUILTIN_LED | GPIO2
D5 | IO, SCK | GPIO14
D6 | IO, MISO | GPIO12
D7 | IO, MOSI | GPIO13
D8 | IO, 10k Pull-down, SS | GPIO15
G | Ground | GND
5V | 5V | -
3V3 | 3.3V | 3.3V
RST | Reset | RST

RX/TX are also used by the USB-to-Serial converter. D0 is very limited (no interrupt/pwm/I2C/one-wire support), D8 has to be in a certain state on boot up, so connecting something to this pin may be a bad idea.
We use the following pins:
D1, D2: software serial for the SDS011
D3, D4: all I2C components
D5, D6: PWM input PPD42NS, second software serial for GPS
D7: DHT22

I2C is a serial bus, so you can connect more than one device in a row ;-). The supported LCDs and OLEDs are I2C devices.

Naesstrom, 2018-02-24:
Yeah I know about the I2C functions, that's why I was planning on adding 3 headers for I2C. I'll have to read up on the ws2812b if they need any of the things that isn't supported on D0 since that seems to be the best bet!
or... change the layout to use a nodeMCU instead since they have pins enough?

Naesstrom, 2018-02-24:
Updated the PCB to fix the vias and added 3 5v I2C connectors to
[image]

dokape, 2018-02-26:
The I2C: not every I2C modul has the same sequence  SDA-SCL-GND-VCC
Sometimes the SDA/SCL is crossed. 
Perhaps you should provide a cross / Jumper for the left/right four I2C placed to change sequence.
Check the BME280, HTU21D, OLED-Display ...

Naesstrom, 2018-03-01:
Hmm, the right I2C's are powered by 5V, not sure if I should remove that and switch the SDA/SCL as you suggest... My personal plan was just to use dupont cables and solder header pins to the board.
Meanwhile, I've found this cute little I2C led (https://www.sparkfun.com/products/8579) that should work, would be great if the code had built-in support for changing the led based on particle count.

Naesstrom, 2018-03-14:
I ordered a set of the ones above and they turned out quite good:
[image]
[image]

Noticed tough that the labels for the SDS sensor got hidden beneath the connector and that I forgot to add labels to the DHT22... So here's the second revision (or rather 1000th)
Added 2 more I2C connectors with the pins mirrored to fit a BME280 directly soldered to them. Also added space to add a 4.7kOhm beneath the Wemos.
6mm spacers and it was a perfect fit at the back of the SDS sensor
[image]

Just thinking now if I should make it  a few millimeters more narrow so that It will fit in the sewage pipes with the sensor...?

dokape, 2018-03-14:
looks pretty nice. Instead of posts I would consider using sockets for the I2C bus.
You could use a good old jumper for chosing 3V/5V. I've using this on my test-board.

Nordrunner01, 2018-03-14:
Hello.
Geat work!!! thumbsup

Where to buy? pcb.io like to share?

Best regards

Naesstrom, 2018-03-16:
Thanks for the idea about a jumper @dokape I'll add that to a future revision, would you suggest one for all I2C connections or split them 3by3? Reason I'm using pins is that I was planning on making it simple for me and use female-female cables the first tests 😸 

@Nordrunner01 I've added all the files to a repo instead https://github.com/Naesstrom/airrohr-PCB

Naesstrom, 2018-03-17:
Awh crap... noticed today when I connected it up that I had the RX/TX reversed, no biggie since I could just switch the positions in the cable but I just ordered 10 more cards

miskaknapek, 2018-03-17:
Well done persevering with such enthusiasm!
Maybe the 'hiccked up' cards can become collector's items ;)

/Miska

Adorfer, 2018-03-17:
1) it could be easily converted to a single layer PCB, with a minimum of 2-3 wire bridges/0ohm-resistors.
2) if somebody would go into bulk production: putting the ESP8266 directly on the main PCB and use an external programmer would cut cost. You just would have to look for different 3.3V supply.

Naesstrom, 2018-03-18:
@miskaknapek I can't just be spamming our FB group all the time, need to distribute it evenly! 😄 

@Adorfer indeed, not sure what the cost benefit would be besides nice to be able to do them at home! Ordering them from a chinese sweat-shop is good enough for me! But I like the idea of adding the ESP directly to the bord... guess adding a FTDI connector on one side and then a voltage divider is easy enough!
My original plan was to place the SDS011 connectors at the exact same place as on the sensor to be able to replace the cable with a direct bridge with pins!

dirkmueller, 2019-11-28:
Thanks for sharing! Is there an open issue left?

DeeKey, 2019-12-30:
Please add DocumentationNeeded tag (or something like that) to add it to documentation. Great work!

ricki-z, 2019-12-30:
We don't change this issue as we are working on an own design (including connectors for noise and SPS30).
This issue is just a reminder.

DeeKey, 2020-01-06:
Here is another opensource realization of HW
https://nettigo.eu/products/nettigo-air-monitor-pcb-0-3-2

Adorfer, 2020-01-07:
@DeeKey intersting detail:  nettigo.eu offers an integrated heating chamber!
"Without heating and control element in RH higher than 70% You can observe condensation. With this process water and dust particles combine making more and bigger objects. Optical sensors can return readings few times higher when condensation phenomenon is present."

it's a bit off topic, but definitivly a thing to look into: 
[image]