opendata-stuttgart/sensors-software #187: sensor: benzene and sulfur

tho-wa, 2018-03-04:
Because of air quality issues in the ore mountain region I would like to know if its also possible to measure benzene (benzol) and sulfur (schwefel) by gas sensor?

Adorfer, 2018-03-05:
it is possible.
But the downsize is that all those sensors in the affordable price range 1) need a lot of power (due to some platinum wire heated several hundred degrees...) and 2) do not last for years, but more for months. 
So even if you can get them for 50€, it will add up after time.

Adorfer, 2018-03-14:
if it would bee "RS232": There are some I2C-IO-externders available (some even with dedicated RS232 on it), Or other I/O-stuff. Off course additional programming needed... and perhaps not suitable for 115200bps... but for 9k6 it will work. 
Like: https://www.sparkfun.com/products/13601