opendata-stuttgart/sensors-software #188: Problem with ISP dynamic IP

DancobMKD, 2018-03-06:
Every time when the ISP router receive IP change (usually on interval at 12th) the sensor
loses connection with Madavi and Openstreetmap.
I need to restart the router so data again to be sent to these tho providers.

Any solution?

I'm on Firmware version: NRZ-2017-099.

ricki-z, 2018-03-06:
This seems like a routing problem. But this is caused by the router software. Please ask your ISP for an update or a new router.

DancobMKD, 2018-03-07:
Is there a option on the software (Firmware) to use DDNS (DynDNS)

ricki-z, 2018-03-09:
DynDNS won't solve the problem. DynDNS will assign a domain to your ISP IP and thereby to your router.

DancobMKD, 2018-03-10:
I think i solved the problem. I entered Google DNS servers in the router, So far no problems.
Thanks

dokape, 2018-03-14:
Nice to hear you solved the issue. 
The Airrohr only sends data from inner network to the outer internet.
Therefor it is not necessary to have a static IP or a DDNS address.
There is no need to access from internet direct to the airrohr. 
So it seems, your ISP/Router Setting to the ISP-DNS has some problem with accessing to madavi after daily update. Using the google-DNS often solves problem.

If your problem seems to be solved please close the issue.

closed 2018-03-15 by DancobMKD