opendata-stuttgart/sensors-software #190: no Oled display's temperature and humidity data with Htu21d

stewecar, 2018-03-12:
Hello,
maybe is a known issue but i don't see any data in this particular combination.
Thanks!
stefano

ricki-z, 2018-03-14:
Displaying the values is work in progress. There will be some major changes with the next version.
Version NRZ-2017-099 shows only values from DHT and BME/BMP sensors.

stewecar, 2018-03-14:
Thanks a lot for the reply.
If possibile i think it could be useful, for the htu21d, to show only a decimal in the temp and humidity values.
Thanks again!
Stefano

dokape, 2018-03-26:
The next firmware displays the HTU21D Data on oled. The firmware will display several screens rotating every some seconds to display a lot of imformation. It will be a big improvement. 
Be patient, you will like it.

stewecar, 2018-03-26:
Wow! Thanks a lot for the infos! Cannot wait it to be released :)

ricki-z, 2018-06-03:
@stewecar new firmware was published and should be updated in the next 1 or 2 days. Please close the issue if the display is working.

stewecar, 2018-06-03:
Thanks! Closing it.

closed 2018-06-03 by stewecar