opendata-stuttgart/sensors-software #192: Provide Prometheus Endpoint

mrueg, 2018-03-19:
https://prometheus.io/ is a monitoring system and time-series database scraping metrics from endpoints provided by systems, services and applications.

It would be nice if the airrohr-firmware could provide a metrics endpoint using the prometheus format in addition to its data.json endpoint.

michaelshmitty, 2018-03-19:
I wonder, given the similarities of prometheus.io with influxdb, if it wouldn't make more sense to provide push functionality for prometheus.io in the airrohr-firmware (like it does for influxdb) rather than implementing an endpoint of which to pull the data from.
It may be complex for sensors behind firewalls, in DMZs or internal networks to make an endpoint available.
I'm not familiar with prometheus's infrastructure yet, but I'm basing my comment on what I read in [the the comparison notes of prometheus.io with influxdb](https://prometheus.io/docs/introduction/comparison/#prometheus-vs.-influxdb).

I would welcome this support in the airrohr-firmware and if I can find the time would be interested in helping to implement this. I currently make heavy use of the airrohr-firmware's support for pushing data to an influxdb server.
But I've always felt a bit uncomfortable with the influxdb product and the monolithic company behind it.
From [the prometheus.io pages](https://prometheus.io/docs/introduction/comparison/#prometheus-vs.-influxdb):

> InfluxDB is maintained by a single commercial company following the open-core model, offering premium features like closed-source clustering, hosting and support. Prometheus is a fully open source and independent project, maintained by a number of companies and individuals, some of whom also offer commercial services and support.

I'm not sure how we're doing on memory usage / resource limitations with the current airrohr-firmware but it could be possible to support both services.

mrueg, 2018-03-19:
Well the data.json is already available to pull from, so a second representation in prometheus metrics shouldn't be a problem (I assume)?

This might help a bit: 
https://prometheus.io/docs/introduction/faq/#why-do-you-pull-rather-than-push?
https://prometheus.io/blog/2016/07/23/pull-does-not-scale-or-does-it/

There's the https://github.com/prometheus/pushgateway available, if push functionality to prometheus is desired.

ricki-z, 2018-03-20:
Please use the 'Own API' feature for this. You can write a mapper and forward data from there to prometheus.io. Any new endpoint or push functionality needs place we need for other features. See the discussion about MQTT.

michaelshmitty, 2018-03-20:
> Please use the 'Own API' feature for this.

Ah yes, completely forgot about that "catchall" feature! Thanks Rajko.

dominikschulz, 2018-03-22:
It would be really nice to have a simple text based prometheus endpoint or at least a proper JSON output (e.g. unquoted numerals) with all the necessary information (e.g. sensor ID) included.

mrueg, 2018-03-23:
After thinking a bit about it, would a prometheus endpoint similarly to the data.json endpoint require the airrohr to be permanently connected to the wifi? Is this the case or does it shut it down wifi between sending data?

ricki-z, 2018-03-25:
The airrohr should be permanently connected to the wifi. We only stop the internal webserver while sending data (otherwise sending via https will crash the firmware).

ricki-z, 2018-03-25:
Most software should be able to work with the quoted numerals. There are connectors for OpenHAB, FHEM, Home Assistant and others and I haven't heard about the need to unquote the numerals.

dominikschulz, 2018-03-27:
Most software can work with quoted numerals and almost all software can made to work with them, that is correct.
But they are a violation of RFC 7159 and may require additional postprocessing.

mrueg, 2018-06-01:
Resolved in https://github.com/opendata-stuttgart/sensors-software/pull/217

closed 2018-06-01 by mrueg

dominikschulz, 2018-06-01:
Wow, thanks a lot :+1: