opendata-stuttgart/sensors-software #195: After power cycle "config file not found"

dxxxm, 2018-03-23:
FW installed as per (Variante1)[https://github.com/opendata-stuttgart/meta/wiki/Firmware-einspielen#variante-1-arduino-ide)
Modified Wifi config in ext_def.h for local wifi ssid and key. No other changes made, i.e. remaining config options untouched.
Activated 'Serial monitor' to monitor things.
Upload ok.  wifi ok. sensor connects, OTA etc ... registers and shows up (line 4453)[https://www.madavi.de/sensor/versions.php]. 'serial monitor' output does not hint any problems.
Any 'reset' attempt leaves sensor in a stalled state. e.g. via local web page or the hardware switch.
After power cycle sensor comes up again but with a 'no config file found'. First tries to join hard coded ssid 'freifunk-disabled', times out, subsequently launches the fall back web server 'feinstaubsensor-xxxx'.
An explicit 'config save' via sensors local page doesn't make a difference. 'Serial monitor' reports the save action. Content (wifi settings) as shown on output are correct.
Tried all sorts of mumbo jumbo. Deleting 'saved config' both variante 1 + 2 and fresh upload does not cure issue.  Any reset, power cycle gives reproducible: 'no config file'
Yday tried to use precompiled FW. No luck either. 
Stuck here. Any hint/idea very much appreciated

ricki-z, 2018-03-25:
Could you please check if you have configured the flash size correctly? Should be "4M (3M SPIFFS)".
And you could try to flash an empty "file system" (https://github.com/opendata-stuttgart/meta/wiki/Firmware-einspielen#l%C3%B6schen-der-konfiguration).

dxxxm, 2018-03-26:
Looks like a wrong flash size setting was the culprit. Flash size setting before was "4M (1M SPIFFS)". Fixing it makes both restart / reboot ops work fine now. Multiple tests done, all went ok. Many thanks.

closed 2018-03-26 by dxxxm