opendata-stuttgart/sensors-software #196: Connecting to shared wifi on an iPhone (iOS) does not work?

peterk, 2018-03-24:
I have put together a mobile sensor (on SDS011) and I want to connect it to shared wifi on my iPhone. However, the connection to shared wifi fails (it looks like the ESP is connecting but immediately disconnects). Has anyone else been successful in connecting the Feinstaub ESP to a internet sharing on an iOS device?

LuchtwachtersDelft, 2018-03-26:
Yes, works for me. 

You probably already got this far but I'll post a recap of the initial setup:

You can set the SSID and password of the iPhone's Personal Hotspot in the sensor's WiFi Settings while it's accessible through a different network. The iPhone's name will be used as its SSID. Both the SSID and the Wi-Fi Password are displayed under Settings, Mobile Data, Personal Hotspot. 

If the sensor isn't connected to a network you can wait until the sensor itself goes into hotspot mode, connect to it through WiFi, then fill in the iPhone's Personal Hotspot name and password.

Once you enable Personal Hotspot, the sensor should connect after a while and a blue bar at top of the screen should appear. Then you connect to the sensor using http://feinstaubsensor-{nodemcu-id}.local in Safari, replacing {nodemcu-id} with the sensor ID that was displayed in the sensor's dashboard. Or you can guess the IP address the iPhone issued, 172.20.10.0 through 172.20.10.15. You can disable Bluetooth by the way. 

If that doesn't work, check if your Personal Hotspot is working with other devices. Also remember some mobile carriers block tethering.

peterk, 2018-03-26:
Thx. Got it working after disabling Bluetooth. Weird but apparently works.

dokape, 2018-03-26:
Nice to hear it works.
Issue seems finished, please close this issue. It's not gone, and we keep the overview. 
Thanks!

closed 2018-03-26 by peterk