opendata-stuttgart/sensors-software #197: Keine Messwerte vom SDS011

TinoM, 2018-03-27:
Hi, ich habe einen Feinstaubsensor mit NodeMCU Lolin v3, DHT22 und SDS011 zusammengebaut und stehe nun vor dem Problem, dass soweit alles funktioniert, nur der SDS011 liefert keine Messwerte.
Die Ansteuerung schein ok, nach (alle) 130 Sekunden läuft der Lüfter an und nach 20 Sekunden geht er wieder aus.
Der serielle Monitor in der Arduino zeigt:
Connecting to xxxxxxxx
........
WiFi connected
IP address: 192.168.0.186
Starting OTA update ...
NodeMCU firmware : NRZ-2017-099
www.madavi.de
/sensor/update/firmware.php
Start reading SDS011 version date
0 - 77 - 4d - 77 .
0 - 40 - 28 - 40 .
0 - 16 - 10 - 16 .
0 - 146 - 92 - 146 .
0 - 250 - fa - 250 .
0 - 170 - aa - 170 .
1 - 197 - c5 - 197 .
2 - 7 - 7 - 7 .
3 - 17 - 11 - 17 .
4 - 8 - 8 - 8 .
5 - 21 - 15 - 21 .
6 - 152 - 98 - 152 .
7 - 34 - 22 - 34 .
8 - 239 - ef - 239 .
Checksum is: 239 - should: 239
9 - 171 - ab - 171 .
SDS version date : 17-8-21
SDS device ID:     9822
End reading SDS011 version date
[update] No Update.

ChipId: 100017
Lese SDS...
Lese DHT...
Sende an luftdaten.info...
Sende an madavi.de...
Auto-Update wird ausgeführt...
Stoppe SDS011...
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
[....]
Das geht konstant so weiter, nur unterbrochen von den API Calls zu luftdaten.info und madavi.de.

[...]
End reading SDS011
Call sensorDHT
Start reading DHT11/22
Humidity    : 42.50%
Temperature : 22.20⸮C
------
End reading DHT11/22
Creating data string:
WLAN signal strength: -62 dBm
------
## Sending to luftdaten.info (SDS): 
No data sent...
## Sending to luftdaten.info (DHT): 
Start connecting to api.luftdaten.info
Requesting URL: /v1/push-sensor-data/
100017
{"software_version": "NRZ-2017-099", "sensordatavalues":[{"value_type":"temperature","value":"22.20"},{"value_type":"humidity","value":"42.50"}]}
HTTP/1.1 403 Forbidden
Date: Tue, 27 Mar 2018 20:40:31 GMT
Server: Apache/2
Allow: POST, OPTIONS
Connection: close
Transfer-Encoding: chunked
Content-Type: application/json

28
{"detail":"Node not found in database."}
0

closing connection
------

End connecting to api.luftdaten.info
## Sending to madavi.de: 
Start connecting to api-rrd.madavi.de
Requesting URL: /data.php
100017
{"software_version": "NRZ-2017-099", "sensordatavalues":[{"value_type":"temperature","value":"22.20"},{"value_type":"humidity","value":"42.50"},{"value_type":"samples","value":"599177"},{"value_type":"min_micro","value":"239"},{"value_type":"max_micro","value":"244183"},{"value_type":"signal","value":"-62"}]}
HTTP/1.1 200 OK
Date: Tue, 27 Mar 2018 20:40:32 GMT
Server: Apache/2
Vary: Accept-Encoding
Connection: close
Transfer-Encoding: chunked
Content-Type: text/html; charset=UTF-8

1b
Sensor: esp8266-100017
ok

0

closing connection
------

End connecting to api-rrd.madavi.de
Time for sending data: 624113
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading...
[...]

Wäre für Tipps zum weiteren Debugging dankbar.

ricki-z, 2018-03-28:
Könntest du den Teil nach dem Senden an die APIs schicken? Ich würde gern sehen, was das Auslesen des SDS011 so ergibt. Dort scheint ja der Fehler zu liegen.

TinoM, 2018-03-29:
Ok, ich habe nochmal den SDS011 deaktiviert, neu gestartet (dann lief es einige Minuten halt ohne jegliche SDS-bezogene Logmeldungen), dann habe ich den SDS011 in der Konfig wieder aktiviert und neu gestartet.

Aktuell dann ein Auszug aus dem Log:
`Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorDHT
Start reading DHT11/22
Humidity    : 43.30%
Temperature : 21.40⸮C
------
End reading DHT11/22
Creating data string:
WLAN signal strength: -47 dBm
------
## Sending to luftdaten.info (SDS): 
No data sent...
## Sending to luftdaten.info (DHT): 
Start connecting to api.luftdaten.info
Requesting URL: /v1/push-sensor-data/
100017
{"software_version": "NRZ-2017-099", "sensordatavalues":[{"value_type":"temperature","value":"21.40"},{"value_type":"humidity","value":"43.30"}]}
HTTP/1.1 201 Created
Date: Thu, 29 Mar 2018 12:06:31 GMT
Server: Apache/2
Allow: POST, OPTIONS
Connection: close
Transfer-Encoding: chunked
Content-Type: application/json

80
{"sensor":11744,"timestamp":"2018-03-29T12:06:31.885553","sensordatavalues":[{"sensordata":990625390},{"sensordata":990625390}]}
0

closing connection
------

End connecting to api.luftdaten.info
## Sending to madavi.de: 
Start connecting to api-rrd.madavi.de
Requesting URL: /data.php
100017
{"software_version": "NRZ-2017-099", "sensordatavalues":[{"value_type":"temperature","value":"21.40"},{"value_type":"humidity","value":"43.30"},{"value_type":"samples","value":"625092"},{"value_type":"min_micro","value":"230"},{"value_type":"max_micro","value":"244149"},{"value_type":"signal","value":"-47"}]}
HTTP/1.1 200 OK
Date: Thu, 29 Mar 2018 12:06:32 GMT
Server: Apache/2
Vary: Accept-Encoding
Connection: close
Transfer-Encoding: chunked
Content-Type: text/html; charset=UTF-8

1b
Sensor: esp8266-100017
ok

0

closing connection
------

End connecting to api-rrd.madavi.de
Time for sending data: 920057
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
End reading SDS011
Call sensorSDS
Start reading SDS011
`

ricki-z, 2018-04-08:
Die Modi werden jetzt beim Sensorstart für alle Sensoren richtig gesetzt. Leider scheint die aktuelle Version der esp8366 Erweiterung ein WiFi-Problem zu haben. Daher kann ein Update der Firmware-Binaries noch etwas dauern.

cfelicio, 2018-04-27:
Im having the same issue, do you have any instructions on how to switch the SDS011 out of query mode?

ricki-z, 2018-12-09:
Wir verzichten aus Speichergründen auf das Auslesen der Antwort durch den SDS011. Es scheint aber ab und zu vorzukommen, das die Kommandos nicht wirklich ankommen, wenn der SDS011 gerade Daten schickt (was in deinem Fall nicht vorkommen sollte) oder aber noch nicht komplett gestartet ist nach einem Start-Kommando. Wir werden da wohl zur Sicherheit in Zukunft die Kommandos zweimal schicken mit einer kleinen Pause dazwischen.