opendata-stuttgart/sensors-software #199: BME280 sensor together with a OLED display ?

Schnellzug, 2018-04-02:
I want to have a BME280 sensor to work together with a OLED display.
Unfurtunaly they are using the same pins.

BMP180 / BMP280 / BME280 (I2C)
-    VCC -> Pin 3V3
-    GND -> Pin GND
-    SCL -> Pin D4 (GPIO2)
-    SDA -> Pin D3 (GPIO0)

OLED displays with SSD1306 (I2C, 128x64 pixels)
    VCC -> Pin VU
    GND -> Pin GND
    SCL -> Pin D4 (GPIO2)
    SDA -> Pin D3 (GPIO0)

Is it possible to modify the firmware so that i can connect BME280 to other (until now) not used pins?
e.g.:
-    SCL -> Pin D10 (GPIO3)
-    SDA -> Pin D9 (GPIO1)

According to the library of the BME280 it is possible:
- SDA/SCL default to pins 4 & 5 but any two pins can be assigned as SDA/SCL using Wire.begin(SDA,SCL) 
#define BME280_PIN_SCL D10
#define BME280_PIN_SDA D9

ricki-z, 2018-04-03:
The BME280 and the display are connected via I2C. This is a serial bus. You can connect the components one after the after to the same pins. You only have to look for the needed power connector (3.3V or 5V/VU) So we don't need more or other pins.

dokape, 2018-04-03:
This works fine together. 
The profit of the serial bus I2C is to use less pins. The connected devices have different internal IDs, so the firmware can read/send the data to each device connected to the bus separately by using the ID of the device. 
Less wiring - more sensors. 

Did it work at your sensor?
 
In actual firmware 099 the displays will maybe not show every value.
With the upcoming firmware 100 the use of display will be improved a lot.

Schnellzug, 2018-04-03:
Thank you for the answer.
I will test it as soon as i will have my OLED.

P.S.:
Where can i download the compiled firmware 100 beta  (bin-file)?

I am a newbee with Arduino-IDE.
I downloaded the source files and all libaries.
When i compile i get the error message:

exit status 1
stray '\302' in program

ricki-z, 2018-04-08:
There are no publicly available binaries for beta version by now.
To the exit status: You need to download the source files via the 'raw' button. The error is caused by compiling the html files.

Schnellzug, 2018-04-08:
Thank you !
The compiling is working now.

closed 2018-04-08 by Schnellzug