opendata-stuttgart/sensors-software #200: Sensor data won't get uploaded

kristigu, 2018-04-06:
Hello everyone,

A few days ago I installed my measurement station with the SDS011, DHT22 and ESP8266.

Initially everything seemed to be going fine but after a while the data would first not get regularly uploaded, and later wouldn't get uploaded at all. https://www.madavi.de/sensor/graph.php?sensor=esp8266-1846195-sds011

However if I access the sensor locally on it's IP 192.168.1.104, the data is regularly updated every three minutes.

I thought it could be a connectivity issue since my modem was located a bit far and the signal was rather poor (40%). So yesterday I installed a new network from a router very close to the sensor and configured it again but the problem persists. 

Last night I managed to get a couple of datasets uploaded fromn1AM to a

DwaineGarden, 2018-04-06:
Go into configuration and set the debug level higher.  With the Arduino console you can see the network connection fail or not.  Also if there is a crash

kristigu, 2018-04-06:
Can you please be more specific on how can I check on Arduino? Also, do I need to connect the boars via USB to the computer?

ricki-z, 2018-04-07:
How is the sensor 'connected' to your wifi network? Is there an IP address shown after start that belongs to your home network? Is there anything blocked on your router?
The sensor can't connect to 2 different domains. So there may be a DNS problem or a connection problem.

kristigu, 2018-04-07:
Indeed it seems like my internet connection is causing the problem. Weird however that it was ok the first two or three days.

I happened to have a mobile data usb key from a mobile network provider and configured the sensor with it. It has been working with no problems for the past 2.5 hours. Let's hope this fixes it!

ricki-z, 2018-06-03:
@kristigu any news? can we close this issue?

henare, 2018-07-11:
Hi everyone :wave: This is my first comment on this project - thanks so much for making it! :revolving_hearts: 

I've recently built my first sensor and I've been having the same issue. You can see spotty little outages on the charts, e.g. https://www.madavi.de/sensor/graph.php?sensor=esp8266-818835-dht

I've been documenting the debugging process on this issue: https://github.com/henare/air_quality_monitor/issues/9

It's weird because I've seen the device drop off my WiFi completely but I've also observed it being connected to WiFi, uploading to a local IP address OK, but not uploading to a remote server using a DNS name.

I'd like some suggestions from others on this issue about what I can do to debug it further. Turning up the debug level and monitoring it over serial didn't reveal any insights (similar to @kristigu's [comment above](https://github.com/opendata-stuttgart/sensors-software/issues/200#issuecomment-379481385)).

My next step is to connect it to my phone's WiFi to completely remove my home network's WiFi as a possible problem. However no other devices have a problem with my home WiFi so even if that tests OK I'd still say the device has a problem.

Thanks in advance for any help :bouquet:

michaelshmitty, 2018-07-11:
> Hi everyone 👋 This is my first comment on this project - thanks so much for making it! 💞
> I've been documenting the debugging process on this issue: henare/air_quality_monitor#9

Definitely check and / or test with different quality power supplies.
High frequency noise, under voltage can cause unexpected behavior within these micro controllers.

Good you’re taking a methodical debug approach. Change only one parameter (power supply, other WiFi network, ...) at the time and test again. Otherwise you risk chasing ghosts.
Good luck and let us know if you figure out what the problem was.
Also keep in mind that the problem may be at the remote end or a combination of problems you have no control over.

Cheers!

dynand, 2019-01-05:
Looks very familiar to my sensor. I notice that the data gets very spotty during the night but during daytime everything seems fine.
I use an Apple power supply in the bike shed so environmental temperature might be a problem. I’ll put a USB Voltage and current monitor in the USBline to see if that gives any clues.

Just a suggestion: can we get a WiFi connection strength in graph at madavi.de? Just to see if that causes problems?

Would adding a condensator in let’s say mF range help stabilise the 5V and thus solve interference?

lodagro, 2019-01-05:
@dynand Wifi signal strength is available at madavi

`https://www.madavi.de/sensor/signal.php?sensor=esp8266-<esp-id>`

dynand, 2019-01-05:
Interesting, I only see sds-011 and dht sensors (6893890). Is the graph available but not listed? Or am I simply missing something?

Dynand

dynand, 2019-01-05:
Thanks! Found it at https://www.madavi.de/sensor/signal.php?sensor=esp8266-6893890

Gives me a clue wether the problem occurs in the sensor or in the 8266 or the connection

ricki-z, 2019-04-28:
Can we close this issue?

kristigu, 2019-04-28:
Yes, for me it's fine to close!

*Kristi Gusho*
+32495157946
Drietorekenstraat 12,
9000 Gent,
Ghent, Belgium

closed 2019-04-28 by ricki-z