opendata-stuttgart/sensors-software #201: Firmware: NRZ-2017-099, Software stays in configuration mode and won't start measuring

dl7jp, 2018-04-10:
Hi, I hope I am right here with this... firstly, thanks for the great work! 

I have successfully built 4 or 5 sensors and I am now trying to set up a PM-Sensor with a local LCD display and no WLAN access. All works fine when it's connected it to the configured WLAN, it uploads data, etc.
However, if it can't find the configured access point it remains in config-Mode and spans the configuration WLAN AP. Timeout is left at 600 sec., but after 20 mins the sensor is still in this configuration mode. I verified this with two different boards, both fail to go into the measurement cycle after config times out as the flowchart at the bottom of https://github.com/opendata-stuttgart/meta/wiki/Konfiguration-der-Sensoren suggests.

dokape, 2018-04-12:
@dl7jp 
The display at version -099 is just simple.
The -100 will improve the displayed data.
I'm not shure if the actual code runs into measurement after wating 600sec on the configuration mode. 
It has not been designed or tested as stand-alone-sensor for only showing data on a display. 

@ricki-z 
At version -099 and -100 it looks like the code does not provide a mesurement without connection to WIFI. 
Are I'm right that the code does check Wifi.status and displays the config page, It does count downs but is not going to a measurement mode as stand-alone sensor only with display?

Does it make sense to provide a stand-alone-solution?
pro: 
- It could be interesting for companies / shops / NGO to show data lokal.
- make more people interesting for the sensor and measurements
contra:
- This project should provide data for everyone (by design, opendata)
- how to protect the access of sensor in startup/configuration. Is basic auth enough?

dl7jp, 2018-04-12:
> I'm not shure if the actual code runs into measurement after wating 600sec on the configuration mode.
It does not, the timeout seems to be ignored.

There are locations where data upload via WLAN is not an option, so a stand-alone sensor with a local display is the only choice. Imho, going into measurement mode after config timeout would be fine: You can then set the timeout to 0 and update the firmware via USB if needed.

dl7jp, 2018-04-12:
I looked into the code, it seems the Boolean config_needs_write causes the while-loop in wifiConfig() to run forever. I am not sure if I will find time to dig into it in foreseeable time, so if anyone cares to fix it: Drop me a line and I will tell you where this sensor is supposed to be deployed :-).

pathmapper, 2018-04-13:
@dl7jp 

> There are locations where data upload via WLAN is not an option, so a stand-alone sensor with a local display is the only choice. Imho, going into measurement mode after config timeout would be fine

There is a workaround if you would have a temporary WLAN available, e.g. using a phone as a WLAN hotspot:
You could connect the standalone sensor to the temporary WLAN and wait for the begin of the measurements. The sensor will continue with the measurements and display the values on the display even if the WLAN is not available anymore.

dl7jp, 2018-04-13:
@pathmapper thanks a bunch, nice hack :-)

dl7jp, 2018-04-13:
I had a closer look at the code for NRZ-2017-100: contrary to what I wrote above config_needs_write  is not the problem. The firmware will restart after the Wifi-Configuration times out rather than go into measurement mode. Deleting "restart_needed = true;" at the end of wifiConfig() prevents this. I am not sure if this breaks anything else, so don't go for it unless you only want a stand-alone version. 
For the moment the issue is solved for me, thanks!

closed 2018-04-28 by dl7jp