opendata-stuttgart/sensors-software #204: Double quote in WLAN password does not work

doreveng, 2018-04-17:
If the WLAN password contains ASCII 0x22 (") it leads to:
> opened config file...
failed to load json config
Starte Webserver... 0.0.0.0

then falls back trying Freifunk-disabled ...
At least all 7bit ASCII printable chars should be supported for the WLAN password.
Also helpful would be a hint about which chars are allowed in WLAN password at:
[https://luftdaten.info/feinstaubsensor-bauen/#feinstaubsensor-konfiguration](url)

Adorfer, 2018-04-19:
BTW: I would like to use "Freifunk" (instead of "Freifunk-disabled") as default, since it's likely that it might be existing somewhere (and working). 
But anyhow: Escaping of chars should be done for PW, otherwise injections might be very likely possible....

ricki-z, 2018-07-11:
This should work with actual beta.

ricki-z, 2018-08-09:
Should work with new actual release NRZ-2018-105.

ricki-z, 2018-09-10:
@doreveng this should work now. Can we close the issue?

closed 2019-01-23 by ricki-z