opendata-stuttgart/sensors-software #205: Compiling with newest library [4.0.0] of [ESP8266 and ESP32 Oled driver for SSD1306 display] does not work

charliesz, 2018-04-17:
Hey guys,
just want to let you know, that compiling with newest library [4.0.0] of [ESP8266 and ESP32 Oled driver for SSD1306 display] does not work. Took me a while to figure that out. I know in the readme file are tested version numbers, but keeping the libraries up to date shouldn't do harm.
Cheers 
charlie

Adorfer, 2018-04-18:
If you know people who like to contribute to the code and the maintenance (regression testing, debugging): PRs welcome. 

Yes, this project would really benefit from more developers. There is so much which could/should be done. but at the moment the bottleneck is the time budget of the people involved who are at the level to deal with these issues. (Just my 2 cents)

ricki-z, 2018-06-09:
@charliesz Compiling with the libraries mentioned in README.md should work now. There were some major problems with the new ESP8266 library 2.4.1. So we had to go back to 2.3.0.

closed 2018-09-29 by ricki-z