opendata-stuttgart/sensors-software #207: WeMos ESP8266 boards WiFi AP bricht ständig ab

offlinehoster, 2018-04-27:
Ich habe ca. 20 von den WeMos ESP8266 Boards. 2 Stück funktionieren einwandfrei. Bei mindestens 3 weiteren kommt der erste initiale Wifi AP nur für wenige Sekunden "hoch" und beendet danach die Verbindung zum Client, sodass keine erste initiale Konfiguration möglich ist.

Ist hier etwas bekannt?

Ich habe es mit unterschiedlichen Versionen aus dem Arduino IDE probiert jedoch ohne Erfolg.
```
~/.arduino15/packages/esp8266/tools/esptool/0.4.8$ ./esptool -vv -cd nodemcu -cb 57600 -ca 0x00000 -cp /dev/ttyUSB0 -cf ../latest_de.bin 
esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 57600
	setting address from 0x00000000 to 0x00000000
	setting port from /dev/ttyUSB0 to /dev/ttyUSB0
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/ttyUSB0 at 57600
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
Uploading 459776 bytes from to flash at 0x00000000
	erasing flash
	size: 070400 address: 000000
	first_sector_index: 0
	total_sector_count: 113
	head_sector_count: 16
	adjusted_sector_count: 97
	erase_size: 061000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting timeout 15000
	setting character timeout 150
	done
	setting timeout 1
	setting character timeout 1
	done
	espcomm_send_command: receiving 2 bytes of data
	writing flash
.................................................................................................................................................................................................................................................................................................................................................................................................................................................................
starting app without reboot
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
closing bootloader

```

ricki-z, 2018-04-29:
Für die Konfiguration ein USB-Netzteil benutzen. Die USB-Anschlüsse mancher Computer begrenzen den Strom auf 100 mA, dann ist die WLAN-Verbindung des ESP instabil.

tom-r, 2020-02-07:
Ich hatte grad so ein ähnliches, eigentlich unerklärliches Phänomen mit dem normalen ESP8266.
Wenn Du selbst compilierst, pack mal eine Zeile "delay(1)" oben in die loop() Funktion. 
Einfach nach der Zeile
...
send_now = msSince(starttime) > cfg::sending_intervall_ms;
**delay(1);**
...
Seither geht's bei mir ...
Einen Versuch ist es wert ...
Gruß,
Thomas