opendata-stuttgart/sensors-software #214: [May 26] no data from any of the stations

nschloe, 2018-05-26:
As of now, no sensor sends data upstream; pick any in https://maps.luftdaten.info/#10/48.2909/8.9730.

[image]

From what I can tell in my own station, the sensor itself if working fine.

ricki-z, 2018-05-26:
Not sending data upstream can't be true. There are values (5 and 5 for ID 5894) shown in the table.
The server was restarted for a security update (Meltdown + Spectre patches). But the grafana instance didn't came up after this. Now the graphs should be shown again.

nschloe, 2018-05-26:
Indeed they are. Thanks for the quick fix! Btw, I'm assuming the jagged data in

[image]

is also due to some server problems? (It's jagged for all sensors in the map at the same times.)

closed 2018-05-26 by nschloe

ricki-z, 2018-05-26:
This should also be fixed now.

nschloe, 2018-05-28:
@ricki-z Jagged data still occurs now; e.g., Sunday evening https://api.luftdaten.info/grafana/render/dashboard-solo/db/single-sensor-view?orgId=1&panelId=2&width=800&height=500&tz=UTC%2B02%3A00&var-node=12338.