opendata-stuttgart/sensors-software #219: HTU21D führt zum crash

bernemer, 2018-06-04:
Hallo,
wenn ich den HTU21D Sensor aktiviere hängt sich der NodeMCU beim ersten Leseversuch der Daten auf.
Ein erneutes ansprechen der IP ist erst nach Neustart des NodeMCU möglich. (Dann nur bis zum ersten Leseversuch)
Das macht er sowohl wenn ich den Sensor angeschlossen habe als auch wenn er gar nicht angeschlossen ist.
Wenn er das nur bei angeschlossenem Sensor machen würde, würde ich auf den Sensor tippen. Aber auch ohne Sensor .... .

Sonst sind keine Sensoren angeschlossen.
Firmware 103
Lolin NodeMCU V3
Direkt die BIN geflasht.
VIN an 3V
GND an GND
SCL an D4
SDA an D3

Ist das bekannt? 
Habe ich trotz Suche eine Information übersehen?

BME280, BMP280, OLED funtionieren einzeln oder zusammen übrigens einwandfrei.

Grüße aus Frankfurt

Marcus

PS: Tolles Projekt

ricki-z, 2018-06-04:
Hast du die NodeMCU zwischendurch vom Strom getrennt? Die NodeMCUs haben ein Problem mit dem 'Softreset' der Arduino IDE nach dem Flashen.
Ich selbst habe hier mit einem angeschlossenen und aktivierten HTU21D getestet.

bernemer, 2018-06-04:
Hallo ricki-z,

Danke für die schnelle Rückmeldung.
Sowohl Reset als auch Stromtrennung bringen keinen Unterschied.
Habe heute die aktuelle BIN von der Luftdaten Seite auf einen neuen NodeMcu über einen Windows Rechner geflasht. 
Also nicht über die Arduino erst kompailiert und damit geflasht. 
Eingeschaltet ohne Sensoren, fürs Wlan konfiguriert, alles APIs abgeschaltet, alles Sensoren abgewählt, 15 sec. Messintervall eingestellt, gespeichert.  Neu gestartet. Alles ok.
Dann HTU21D als Sensor gewählt und gespeichert. HTU21D angeschlossen. Strom drann. Geht bis die erste Messung geholt wird. Dann geht nix mehr.
Dann habe ich den Sensor entfernt und den NodeMcu wieder eingeschaltet. Läuft bis zur ersten Messung dann hängt er wieder bis ich Reset oder Strom ab und dran mache. 

Habe es mit 2 NodeMcu aber dem gleichen Sensor probiert.

Wenn ich nur einen BME aktiviere aber nicht anschließe hängt er sich nicht auf, zeigt nur keine Daten an da ja kein sensor dran ist.

Wie ist die richtige Reihenfolge. Erst Sensor auswählen dann anschließen ohne Strom oder umgekehrt?

Sehe ich das Richtig das in der BIN schon alle richtigen Bibliotheken enthalten sind?
Meine Prgrammierkenntnisse sind ca 25 Jahre her und ich bin mich gerade erst am einarbeiten.

Was kann ich machen um diesem Fehler auf die schliche zu kommen?
Kann ich mit der Arduino Software was mittrace?
Soll ich mal mit der Arduino IDE die .ino mal direkt flashen und nicht die Bin von der luftdatenseite nehmen?

Einen neuen DHT21D hab ich vorsichtshalber schon mal bestellt.

Gruß
Marcus

ricki-z, 2018-06-04:
Hallo Markus,
die 'richtige' Reihenfolge ist ohne Strom anschließen, dann den Sensor in der Konfiguration aktivieren.
Ohne angeschlossenen HTU21D könnte der Absturz durch den Treiber verursacht werden, der vielleicht mit der Nicht-Antwort nicht zurecht kommt. Und sollte der HTU21D einen Defekt haben und deshalb nicht anworten, wäre das dann mit angeschlossenem HTU21D das gleiche Resultat.

Grundsätzlich sollte wenn möglich die fertige BIN geflasht werden. Dort sind alle Bibliotheken in Versionen enthalten, die wir getestet haben. Z.B. ist die aktuelle Version der DHT22 Bibliothek nicht für den ESP8266 brauchbar. Im BIN ist eine Version enthalten, die funktionieren sollte.

bernemer, 2018-06-04:
Hallo ricki-z,

danke für die Hinweise. Ich werde es morgen nochmal testen und dann berichten.
Wenns in der richtigen Reihenfoge auch nicht klappt dann warte ich auf den neuen Sensor und teste dann nochmal.

Ihr macht ihr ein tolles Projekt und bekommt bald einen weiteren Sensor für die Karte in Frankfurt/Main gemeldet.
Noch sind einige Bauteile für die finale Stromversorgung im Zulauf. Daher hab ich ihn noch nicht gemeldet.

Danke 

Gruß

Marcus

bernemer, 2018-06-05:
Hallo ricki-z,

habe es jetzt in der richtigen Reihenfolge angeschlossen.
Gleiches Fehlerbild. Auch nochmal neu geflasht und alles angeschlossen und dann erst aktiviert; kein Erfolg.
Ich denke ich warte auf den neuen Sensor und teste dann nochmal. 
Ich für den Issue erst mal schließen und bei Bedarf einen neuen machen.

Ein Frage hätte ich noch: Werden beim LCD2004 alle 4 Zeilen so wie beim OLED Display angezeigt?
Bei den Bildern der Entwicklung der Displayanzeige waren beim LED2004 "nur" 2 Zeilen angezeigt.

Danke für den support

Gruß

Marcus

ricki-z, 2018-06-05:
Die aktuelle Firmware nutzt auf dem LCD2004 auch 4 Zeilen. Der Screenshot war noch aus Version NRZ-2017-099.

bernemer, 2018-06-05:
Danke für die Info.

Bin gespannt auf die künftigen Weiterentwicklungen.

closed 2018-06-05 by bernemer

bernemer, 2018-07-06:
Zur Info,

es lag an einem defekten HTU21D.
Mit dem neuen funtioniert es problemlos.