opendata-stuttgart/sensors-software #220: No "signal" in influx API data

sensor-freak, 2018-06-05:
Hi,

I just noticed that the data value "signal" is not included in the (additional) InfluxDB API. Is there a valid reason for this?

If not, would someone mind to include the value in the data string? It seems that there are only two adjustments have to be made in function create_influxdb_string():

- remove the term "- 1" in the condition of the for loop
- change the term "- 2" to "- 1" in the if statement where the need for a comma is checked.

Kind regards

ricki-z, 2018-06-09:
The reason: signal value was included only for 'debbuging' wifi problems.
We will include this in the next release .

ricki-z, 2018-07-11:
This should work with actual beta.

ricki-z, 2018-08-21:
@sensor-freak signal should be included in InfluxDB data. Can we close this issue?

sensor-freak, 2018-08-23:
Fixed with commit 00d3d6bfe175897ebff2ee36e8bf804309975f8e, works as expected now.

closed 2018-08-23 by sensor-freak