opendata-stuttgart/sensors-software #224: Space in WIFI SSID won't work

CUnknown, 2018-06-10:
I have a space character in my WIFI SSID and it seems that the sensor will not connect. When creating a guest network without the space, the sensor connects. I found that there is a update for the ESP8266 to support spaces in the WIFI SSID, could you port this fix to the sensor firmware? See https://www.reddit.com/r/esp8266/comments/5pxivk/esp8266_ssid_with_spaces/

YaguraStation, 2018-06-10:
@CUnknown , have you by any chance tested unicode as well?

CUnknown, 2018-06-10:
@YaguraStation Thank you for your fast reply, do you mean replacing the space in the SSID by '\u0020' ?

YaguraStation, 2018-06-10:
Basically I'm talking about emojis - very commonplace in SSIDs these days. :dog:

CUnknown, 2018-06-10:
I know about unicode, but I am not sure how I could enter it in the interface without it being escaped.
I now used '\\0020' and then it connects one time and then restarts (the same behaviour as with the normal space entered in de SSID). Any thoughts?

CUnknown, 2018-06-10:
This is the console output:
`---- Result from Webconfig ----
WLANSSID: TEST\\u0020TEST
----
Reading ...
PPD: 0
SDS: 1
PMS24: 0
PMS32: 0
HPM: 0
DHT: 1
DS18B20: 0
HTU21D: 0
BMP: 0
----
Send to ...
Dusti: 1
Madavi: 1
CSV: 0
----
Autoupdate: 1
Display: 0
LCD 1602: 0
Debug: 3
----
Restart needed ...

WiFi connected
IP address: 0.0.0.0
saving config...
<NOT INCLUDED>
Config written: <NOT INCLUDED>
H⸮⸮ԅ⸮⸮D⸮
?mounting FS...
mounted file system...
reading config file...
opened config file...
parsed json...
Starting Webserver... 0.0.0.0
6
Connecting to TEST\u0020TEST
........................................
Starting WiFiManager
AP ID: Feinstaubsensor-8552623
Password: 
`

Notice the IP adress being 0.0.0.0

I also saw once that it got a valid ip adres for my subnet, but it still restarted.

CUnknown, 2018-06-11:
Maybe building the sketch myself with the newer version of the ESP8266 wifi stack would do the trick, I will try that in a bit...

YaguraStation, 2018-06-11:
Yeah, sorry, didn't want to overflow the thread... but will also attempt to at least reproduce the issue and test emojis sometime this week.

YaguraStation, 2018-06-11:
I could reproduce problems with space in SSID. 

Funnily emoji in SSID works fine, though. 
[image]
[image]

ricki-z, 2018-07-01:
@CUnknown Space should work now (since version NRZ-2018-103). Please close issue if it's okay.

CUnknown, 2018-07-15:
@ricki-z Infortunately, still no luck. After configuring the WIFI network, the console output displays that it required a IP address on the network, but after storing the config, it needs to restart. After the restart, the device is not able to connect tot the WIFI network with a space in the name again.

ricki-z, 2018-08-09:
@CUnknown Should work with new actual release NRZ-2018-105.

ricki-z, 2018-08-09:
Should work with new actual release NRZ-2018-105.

ricki-z, 2018-08-21:
@CUnknown can we close this issue?

RoTahnWaurie, 2018-09-24:
@ricki-z With NRZ-2018-111 this issue occurred for my sensor, too. I installed the sensor  last month with NRZ-2018-107 without any wlan problems. After the automatic update from september 20th it won't connect to my wlan with an whitespace in the SSID. 

Any idea how to fix this - without changing the SSID?

ropeters68, 2018-09-25:
I don't have the issue, last Saturday (22th) I installed version 111 on two new ESP boards. No problem with the connection to my network (with space in SSID). I do have an issue with access to Madavi.de, looks like the website is unavailable?
 
----Origineel Bericht----

RoTahnWaurie, 2018-09-26:
Thanks for the info, but my AP is 2,4Ghz only and was configured to use 802.11n only. The NodeMCU Lolin V3 worked fine for about a month with this setup until update NRZ-2018-111.

closed 2019-04-01 by ricki-z