opendata-stuttgart/sensors-software #225: Interesting Sensor: Lightning Detection

ccoenen, 2018-06-11:


ccoenen, 2018-06-11:
I can't find a good breakout right now, so this might be the end of it. The chip itself sells for ~13$ on Digikey, but without a PCB it's not tinker friendly. I also have no idea how well it performs, I haven't used it myself, I just accidentally found it. If this is out of scope for this project, feel free to just close the ticket.

ccoenen, 2018-06-11:
Related:
http://de.blitzortung.org/live_lightning_maps.php

hakan42, 2018-06-11:
I found that project as well, but they don't seem to be too motivated to expand their coverage. You have to wait multiple years to buy a hardware set from them.

Marwe, 2018-06-15:
I remember you don't need too many stations, but there are multiple projects about it. A commercial one (with Siemens Sensors) is not open, and there are two "competing" free hardware projects. IIRC it should be no big deal to add a station. But since the triangulation works > 500 km there are not too many more stations (in Europe) necessary.