opendata-stuttgart/sensors-software #227: DHT22 and BMP180 together, which one will win in case of temperature?

SunboX, 2018-06-20:
I want to use the DHT22 because I read it has the best accuracy.

For air pressure measurement I want to use the BMP180. But this one also has temperature measurement build in. 

So if I use both together, which one will "win" in case of reporting temperature?

miskaknapek, 2018-06-21:
Btw. I've been comparing results from DHT22 and BME280, and the BME280 is
definitely better with humidity measurements.

SunboX, 2018-06-21:
I guess, luftdaten.info will use the DHT22 temperature measurement. Because I read, that I have to tell that I'm also using the BMP180 so it will be manually added to the database. I guess this manually adding step makes sure, that the DHT22 temperature will be used?

PS: I also own a BME280... so if someone tells me, that it also works fine. I maybe use this one instead of the other both. Saves costs (for future sensors) and also saves some wiring.

miskaknapek, 2018-06-21:
Thanks for the notes André.

I'm looking at the sensor data ( I'm sort of a data analytics person ) and
will do a comparison between the BME280 and DHT22 for humidity data, one of
these days…

Bests /  m

ricki-z, 2018-06-21:
For the published data: all sensor data from registered sensors is published. So if both the DHT22 and the BMP180 are registered the temperature of both will be published. 
Graphs at Madavi.de: graphs are created for all sensors.
Maps: there its a matter of implementation. Most maps (if the show temperature and humidity values) will prefer the temperature of the BMP180. But the may also show both.
Sensor internal webserver: shows all values
Displays (OLED, LCD) connected to the sensor: the order of preference for displayed values is DHT22 DS18B20 -> HTU21D -> BMP180 -> BMP280 ->BME280. So in your case the temperature values of the BMP180 is shown on the display.

SunboX, 2018-06-21:
I read, that the temperature differs by 2°C between those both and that the DHT22 reports the more correct temperature.

Do you also have some time to look into this?

ricki-z, 2018-06-21:
@SunboX How are they positioned? The temperature sensors should be placed as near as possible to one of the openings.

SunboX, 2018-06-21:
I did not investigate both temperatures. It's just what I've read somewhere. My electronics is working, but I have to put everything together now. Maybe today. Thanks for tge hint with the positioning!

ricki-z, 2018-06-21:
The slightly higher temperature was for the BME280 (temp., humidity, pressure). The Arduino library for this sensor sets a continuous operation mode as default. And this leads to the higher temperature. We have changed the mode so that the BME280 is only active when needed.

SunboX, 2018-06-21:
Ok, I installed it now at it's final destination (with DHT22 and BMP180). Both sensors are close to each other and close to the opening of the airRohr:
* https://www.madavi.de/sensor/graph.php?sensor=esp8266-13905634-dht
* https://www.madavi.de/sensor/graph.php?sensor=esp8266-13905634-bmp
* https://www.madavi.de/sensor/graph.php?sensor=esp8266-13905634-sds011

https://opensensemap.org/explore/5b28d3891fef04001bd7dfae

The temperature values are slightly different. But it's ok for me, thanks for your help @ricki-z & @miskaknapek !

@miskaknapek I'm really interested in your evaluation. Can you give me a hint please when you are done? (andre.fiedler@me.com)

closed 2018-06-21 by SunboX