opendata-stuttgart/sensors-software #231: Soft-GPS function request

vossilius, 2018-07-19:
Understand there is the option of adding a GPS sensor. Yet, why not providing a soft-GPS function if a hardware GPS module is missing? Since the Feinstaubsensor has Internet access anyway, you could estimate the geolocation e.g. IP-based and get time and date via NTP query.

Adorfer, 2018-07-19:
doing a wifi-scan of the surrounding is a lot more reliable... a lot of freifunk communities offer this for their routers. Scan the surrounding and hand this BSSID-List to a central server. That one does the handshake with appropriate databases (Mozilla, Wiggle.net etc) and retransmitts the result to those nodes.

vossilius, 2018-07-19:
agree. I put IP-based as an example. Your way would likely be more accurate, too.

ricki-z, 2018-07-19:
The GPS support is very experimental.
Why should we need the geo position for sensors with fixed positions? 'Soft-GPS' isn't reliable enough to get a position that we can work with.
NTP was tested, but all tested implementations were very unstable. It seems like time changes kill some of the functions we use.

vossilius, 2018-07-23:
I am confused. If there is no use case in the first place, as you say, why do you test and implement GPS support as it's in the code? 
To my knowledge accuracy using the method suggested by @Adorfer is a few hundred meters.

Adorfer, 2018-07-23:
I am still confused by the term "Soft-GPS" in the title of this issue. 
Yes, the RS232 used for the GPS so far are SoftSerial-Libraries. And those lack stability or at least tend to break other stuff. A real Hard-RS232 for connecting the GNS would be a lot better and easier. 
But this will probably be a task to be done after migration from ESP8266 to ESP32. 

What we are talking here is WifiSSID-localization, which might help for some users in fixed environments which do not "feel" the need (or forget) to write an email to the gmail-account of ricky.  
off course there is an uncertainty factor of about 100-200m on WifiSSID-localization, but four our goal it should do the trick.

dokape, 2018-08-27:
The hardware GPS via serial interface works with some limitations. As the serial interface implementation seems to be time-critical, the access to the webinterface can produce a bufferoverflow with sensor-restart. 
If you now that limitations, you can use the hardware GPS for doing time limited measurements out there.  perhaps with Data send through the mobile-phone-network via Hotspot for some limited time. 
E.G. to measure dust produced by accidents like fire or industrial cause.  
For fixed sensors, the soft-gps would be not very helpful, you can find on openstreetmap, google maps or Smartphone-GPS much better and easier the exact location.