opendata-stuttgart/sensors-software #233: Potential bug in BME280 code

vossilius, 2018-07-23:
What I try to accomplish is to port your code (NRZ-2018-103) to Wemos D1 mini Pro due to the external antenna it has. I have successfully moved SDS011 and DHT22 to other ports in order to free up the I2C port for the BME280 (ext_def.h). 

The BME280 breakout board I use is not from Adafruit. Using an I2C scanner code it finds the BME280 connected to the I2C pins freed up above.

Using Adafruit's BME280test.ino (modified onto the Wemos D1 mini I2C ports and sensor address 0x76) I can access the sensor and readout data only if I do the modification suggested in Adafruit_BME280.cpp [here](http://davidstein.cz/2017/09/30/arduino-bme280-could-not-find-a-valid-bme280-sensor-solved/). (see 2.1, 2.2; I will come back to this modification further down.)

Firstly, in NRZ-2018-103 I noticed that the I2C pins of the nodeMCU are hardcoded defeating the pin declaration in ext_def.h: 
`Wire.begin(D3, D4);`
So, I changed this into 
`Wire.begin(BME280_PIN_SDA, BME280_PIN_SCL);`
only to find the code 'hangs' after the debug output 
`...`
`Read DHT...`
`Read BME280...`
`Trying BME280 sensor on 76`
Bei 'hang' I mean nothing happens after this output.

From here, I tried two ways:

a) If I don't do the modification above, NRZ-2018-103 does not find the sensor on 0x76 at all (even though the I2C scanner code mentioned above confirms it is there).

b) If I don't command out the `return false;` as per the modification but put `return true;` instead, NRZ-2018-103 DOES NOT HANG after `Trying BME280 sensor on 76` but now only returns zero values when later reading the sensor data. Now, counterchecking Adafruit's BME280test.ino again it only returns zero values, too.

Now I am stuck. To summarize, I can make my BME280 sensor works on the I2C pins of the Wemos D1 mini Pro using the Adafruit library but not in NRZ-2018-103. I can't help but conclude it's some bug in NRZ-2018-103. Or else, any idea how to solve this?

ricki-z, 2018-07-23:
Our firmware checks both I2C addresses 0x76 and 0x77. So there is no need to make changes to the BME280 library (have you read the linked article???).
I have pushed a new beta version with one definition per interface. So now you should be able to change the interfaces in ext_def.h

vossilius, 2018-07-23:
Thanks for your reply. 

Yes, I haven’t changed the bus address in the library for this reason. The change I made is in the .cpp file of the library concerning a sensor chip-ID check Adafruit has implemented (see the link in my initial message). Therefore, my issue is not related to bus address but presumably some issue related to the Adafruit library and your code using it.

Besides, thanks for pushing the beta.

vossilius, 2018-07-24:
Turns out that the odd behaviour I described above is due to the BME280.begin() statement in the initBME280() function never returning because it gets stuck in a while loop within Adafruit's library: 
```
bool Adafruit_BME280::init()
{
    ...
    // if chip is still reading calibration, delay
    while (isReadingCalibration())
          delay(100);
    ...
}
```
I can prevent this by moving the Wire.begin() statement from the setup() function into the initBME280() statement like so:
```
bool initBME280(char addr) {
	debug_out(F("Trying BME280 sensor on "), DEBUG_MIN_INFO, 0);
	debug_out(String(addr, HEX), DEBUG_MIN_INFO, 0);

  Wire.begin(BME280_PIN_SDA, BME280_PIN_SCL);
  
	if (bme280.begin(addr)) { 
       ...
```

I think to change all initBM...() functions accordingly makes a lot of sense anyway in order to use the different pin declarations made for the different sensors defined in ext_def.h.

ricki-z, 2018-07-24:
In the actual beta version there are no different sensor pin definitions. There are only API 'pin' definitions that we need to differentiate the components.
And I don't know why there should any need for more than one Wire.begin(). The library is working with the NodeMCU. And the Wemos D1 Mini is using the same ESP8266 version. I'm using our firmware with a Wemos D1 Mini as a test system. Attached are BME280, BMP180, DHT22, HTU21D, LCD2004, LCD 1602 and an OLED display, all at the same time.
I really don't know what you may have changed on other places in the firmware that the lib won't work as expected.

vossilius, 2018-07-24:
Thanks. Case closed from my end.

closed 2018-07-24 by vossilius

Adorfer, 2018-07-24:
I can confirm that is is working with any BME280 we received so far from ebay and aliexpress. 
The only issue is unreliable BME280. If you have the choice take models with integrate VR and feed it by 5VDC, creating the 3V3 locally.