opendata-stuttgart/sensors-software #234: NRZ-2018-104-B9: InfluxDB configuration doesn't seem to "stick"

arouanet, 2018-07-24:
When running NRZ-2018-104-B9, I'm running into a problem where the configuration options for the checkbox "Send to InfluxDB" (`send2influx`) and for the InfluxDB server name (`host_influx`) don't seem to persist after a restart, and even randomly revert back to an empty value after a while. Other configuration options seem to behave normally as far as I can tell.

I don't know if this is related, but I have also noticed that trying to display the Configuration page (`/config`) after a restart results in a blank page until the first measurement is carried out. After the first measurement, the page displays as expected.

I have looked at the logs, but didn't see any indication that something was going wrong.

Reverting to the current stable version (NRZ-2018-103) makes both problems disappear.

ricki-z, 2018-07-24:
There ws the same error in a comparison as for the opensensemap config. I have changed this and uploaded a new beta version (-B11).

arouanet, 2018-07-24:
Ah yes, that would explain it. It seems to be working as expected, thanks.