opendata-stuttgart/sensors-software #235: Bug: NRZ-2018-104-B11. No config site before first measure

dokape, 2018-07-28:
NRZ-2018-104-B11
Testscenario:
Restart with power off
Sensor Connects wifi. 
Sensor checks updates,
No Update found,
Sensor is reachable with webinterface.
Startseite/HOME:Fine
Values: fine (no values, counting down 145sec)
Config-Page: EMPTY (Bug)

Config Page appears after first measure is done and first values are visible.

Connected: SDS011, DHT22, HTU21D, oled and LCD.

arouanet, 2018-07-28:
I have noticed this as well. I mentioned it in passing in issue #234, but I wasn't sure whether it was related to the configuration bug I was seeing. It looks like it is a separate issue then. As far as I can tell, it was already occurring in -104-B09, but I don't think it was a problem in version -103.

ricki-z, 2018-07-28:
Okay, I can reproduce this on my test system. Funny enough as there shouldn't be any dependencies to the measurements or the actual time ...

ricki-z, 2018-07-30:
It was a memory issue. The config page was too large. What I don't unterstand: the free heap is smaller after sending data (creating strings for last measurements,...). So the blank page should appear after the first measurement if the page is really too large.
But it's working now, so it's only another mystery ...

ricki-z, 2018-08-09:
Should work with new actual release NRZ-2018-105.

ricki-z, 2018-08-21:
@dokape @arouanet can we close this issue?

arouanet, 2018-08-21:
@ricki-z Sure!

dokape, 2018-08-27:
Firmware: NRZ-2018-107 Seems to be fixed. 
Closed.

closed 2018-08-27 by dokape