opendata-stuttgart/sensors-software #241: Sending to own api (file and php/mysql): temperature in all values

bastifix, 2018-08-30:
Hi, I'm sending the data to my own api and save it in a file like in this demo:
https://github.com/opendata-stuttgart/madavi-api/blob/master/data_simple.php

and also in a MySql db with
```
foreach ($schluessel as $key) {
 $index = array_search($key, array_column($sensoren, "value_type"));  # ab PHP 5.5.0
 $zeile .= $sensoren[$index]["value"].",";
 $werte[$i] =$sensoren[$index]["value"];
$i++;
}
```
and
`$eintragen = mysqli_query($db,"INSERT INTO feinstaub_data (s_id,P1,P2,temperature,humidity,GPS_lat,GPS_lon,GPS_height) VALUES ('$id','$werte[0]','$werte[1]', '$werte[2]','$werte[3]','$werte[4]','$werte[5]','$werte[6]')");`

I'm using the newest firmware version, the DHT22 and the Neo6M.
Many data are stored successfully without a problem, but often the temperature value just appears as all value data. Here are some data-lines:

```
2018-08-30 21:50:05,8.90,8.23,22.90,50.10,48.243445,8.105917,291.50,08/30/2018,19:50:04.00,-78
2018-08-30 21:52:31,10.47,8.77,22.70,50.70,48.243453,8.105898,292.30,08/30/2018,19:52:30.00,-78
2018-08-30 21:54:58,8.28,8.68,22.90,50.70,48.243462,8.105895,292.40,08/30/2018,19:54:56.00,-78
2018-08-30 21:57:24,23.10,23.10,23.10,50.20,48.243398,8.105935,285.60,08/30/2018,19:57:23.00,-79
2018-08-30 21:59:50,22.80,22.80,22.80,50.20,48.243410,8.105952,288.30,08/30/2018,19:59:48.00,-78
2018-08-30 22:02:16,8.70,8.00,23.00,50.40,48.243430,8.105935,290.80,08/30/2018,20:02:15.00,-78
2018-08-30 22:04:43,8.50,8.80,23.10,50.60,48.243425,8.105922,291.60,08/30/2018,20:04:42.00,-78
2018-08-30 22:07:41,23.10,23.10,23.10,48.70,48.243435,8.105937,291.40,08/30/2018,20:07:38.00,-77
2018-08-30 22:10:08,23.20,23.20,23.20,50.90,48.243447,8.105947,291.80,08/30/2018,20:10:06.00,-80
2018-08-30 22:12:33,8.47,8.67,22.80,48.70,48.243438,8.105943,290.70,08/30/2018,20:12:32.00,-77
2018-08-30 22:14:59,8.50,8.00,22.60,50.20,48.243452,8.105942,290.70,08/30/2018,20:14:58.00,-77
2018-08-30 22:17:26,22.90,22.90,22.90,50.60,48.243448,8.105950,290.00,08/30/2018,20:17:25.00,-76
2018-08-30 22:19:52,23.00,23.00,23.00,50.00,48.243442,8.105948,290.00,08/30/2018,20:19:50.00,-77
2018-08-30 22:22:18,8.70,8.10,22.70,50.20,48.243438,8.105948,290.90,08/30/2018,20:22:17.00,-78
2018-08-30 22:24:44,8.55,8.95,22.50,50.40,48.243435,8.105945,291.80,08/30/2018,20:24:43.00,-77
2018-08-30 22:27:11,8.65,8.95,22.70,50.80,48.243438,8.105957,290.40,08/30/2018,20:27:10.00,-77
2018-08-30 22:29:37,8.53,7.73,22.80,50.40,48.243445,8.105945,290.80,08/30/2018,20:29:36.00,-77
2018-08-30 22:32:03,8.90,8.10,22.60,50.40,48.243433,8.105943,291.60,08/30/2018,20:32:02.00,-77
2018-08-30 22:34:30,22.80,22.80,22.80,50.70,48.243435,8.105945,291.50,08/30/2018,20:34:28.00,-77
2018-08-30 22:36:56,22.90,22.90,22.90,50.00,48.243443,8.105947,291.10,08/30/2018,20:36:55.00,-79
2018-08-30 22:41:12,8.40,7.90,22.50,50.10,48.243442,8.105942,291.20,08/30/2018,20:41:10.00,-77
2018-08-30 22:43:38,8.77,8.17,22.80,50.30,48.243447,8.105930,291.00,08/30/2018,20:43:37.00,-78
2018-08-30 22:46:04,8.90,8.30,22.90,50.40,48.243450,8.105933,291.20,08/30/2018,20:46:03.00,-78
2018-08-30 22:48:31,8.23,8.60,23.00,51.00,48.243437,8.105938,291.30,08/30/2018,20:48:28.00,-77
```

Nothing changes between the right datas and the only temperature datas. Can anyone imagine what could be wrong there?

Thanks!

ricki-z, 2018-09-03:
Could you change your script to log the JSON that is received?

bastifix, 2018-09-03:
I think the problem is somewhere in the called api url that the data is just missing. I described it here:
https://github.com/opendata-stuttgart/sensors-software/issues/243

ricki-z, 2018-09-03:
Could you post your whole script? I can't find the array $schluessel in data_simple.php ...

Frickelpiet, 2018-10-19:
This should work:

`<?php
#Die Zugangsdaten für die Datenbank werden eingebunden
require_once 'connectdb.php';

#Die Verbindung zur Datenbank wird hergestellt
$pdo = new PDO("mysql:host={$dbhost};dbname={$dbname}",$dbuser,$dbpass);

$schluessel = array("SDS_P1", "SDS_P2", "BME280_temperature", "BME280_humidity", "BME280_pressure");
$json = file_get_contents("php://input");
$daten = json_decode($json, true);
$sensoren = $daten["sensordatavalues"];

foreach ($schluessel as $key) {
 $index = array_search($key, array_column($sensoren, "value_type"));
 $werte[] = $sensoren[$index]["value"];
}

$statement = $pdo->prepare("INSERT INTO finedustsensor (finedust25, finedust10, temperature, humidity, pressure) VALUES (?, ?, ?, ?, ?)");
$statement->execute(array($werte[0], $werte[1], $werte[2], $werte[3], $werte[4] / 100));

#Eine neue id wird vergeben: Die Datenbank trägt automatisch den richtigen Wert ein
$neue_id = $pdo->lastInsertId();

?>`

ricki-z, 2019-11-06:
@bastifix is this problem solved? If yes please close this issue.

dirkmueller, 2019-11-27:
Please reopen if there is an issue remaining.

closed 2019-11-27 by dirkmueller