opendata-stuttgart/sensors-software #247: Sensor software seemingly crashes when API host is slow to respond

arouanet, 2018-09-10:
It seems to me that whenever a host fails to answer in a timely manner, the sensor software crashes and restart rather than timing out and either retrying or moving to the next host.

This has happened to me in the past with the Madavi API when the WiFi signal to my sensor was poor. Moving the WiFi access point closer to the sensor has greatly reduced the number of random restart.
Yesterday, the Luftdaten API was also seemingly either unavailable or slow, and this has resulted in quite a lot of sensor restarts and missing data.

Is this a known issue? Is it worth looking into or is it too complex to fix? I'm happy to have a closer look and try to get logging information if it's useful.

ricki-z, 2018-09-10:
As long as the sensor is restarting and not hanging this shouldn't be a problem. For the databases this is like a missing transmission (same as timeout).
Some of the timeouts are set hard in the esp8288 library, it's not possible to change them.
It seems that some connections also may crash because of the HTTPS connection. Some session keys seem to crash the sensor. But this is also a part we can't change.

arouanet, 2018-09-10:
It seems that the issue with non-configurable timeouts is at least partially fixed? (esp8266/Arduino#3257)
But I have little experience with ESP8266 programming so I could definitely be wrong.

But yes fair enough, I understand this isn't really a problem in practice. Except when you also connect to a custom database or API, as mentioned in #248. 😁

dokape, 2018-09-11:
With NRZ-2018-109 i have a some Crashs/restarts. 
I former version (i think the -103?) did more than 30 000 measuremet cycles. Now i have only about 1000-2000. i have a raspberry pi as local api. There i recognized since 3 days lots of missing Data.

ricki-z, 2018-09-11:
We set HTTPS as default for transmission. But we may add a possibility to disable this (default will be on). The HTTPS connections seem to need too much RAM and the cleanup is too slow.

dirkmueller, 2019-11-22:
is this still an issue with current firmware? can this be closed?

dirkmueller, 2019-12-19:
Closing. don't hesitate to reopen or create a new issue if this persists with new firmware.

closed 2019-12-19 by dirkmueller