opendata-stuttgart/sensors-software #249: Captive Portals

LuchtwachtersDelft, 2018-09-10:
A Captive Portal is a Web page that the user of a public-access network is obliged to view and interact with before access is granted. You may know them from free WiFi hotspots.

Airrohr also has a Captive Portal, but it only works on Android. I am interested in expanding it to iOS. 

A Captive Portal improves the user experience, because users don't have to go to their browser and enter 192.168.4.1 to open the initial config page.

[image]

Every OS has a different way to detect a Captive Portal.

https://success.tanaza.com/s/article/How-Automatic-Detection-of-Captive-Portal-works
https://serverfault.com/questions/679393/captive-portal-popups-the-definitive-guide

A caveat of the Captive Portal is when the configured WLAN becomes unavailable, the sensor will eventually enter AP mode and launch Captive Portal. If WLAN returns, the sensor is still in Captive Portal. See: https://github.com/tzapu/WiFiManager/issues/705

There may also be annoyances with Captive Portals in general.
https://github.com/httpwg/wiki/wiki/Captive-Portals

LuchtwachtersDelft, 2018-09-10:
In [WiFiManager](https://github.com/tzapu/WiFiManager/blob/development/WiFiManager.cpp) it's handled this way:

```
server->onNotFound (std::bind(&WiFiManager::handleNotFound, this));
```

```
void WiFiManager::handleNotFound() {
  if (captivePortal()) return; // If captive portal redirect instead of displaying the page
```

```
/**
 * HTTPD redirector
 * Redirect to captive portal if we got a request for another domain. 
 * Return true in that case so the page handler do not try to handle the request again. 
 */
boolean WiFiManager::captivePortal() {
  DEBUG_WM(DEBUG_DEV,"-> " + server->hostHeader());
  
  if(!_enableCaptivePortal) return false; // skip redirections

  if (!isIp(server->hostHeader())) {
    DEBUG_WM(DEBUG_VERBOSE,F("<- Request redirected to captive portal"));
    server->sendHeader(F("Location"), (String)F("http://") + toStringIp(server->client().localIP()), true);
    server->send ( 302, FPSTR(HTTP_HEAD_CT2), ""); // Empty content inhibits Content-length header so we have to close the socket ourselves.
    server->client().stop(); // Stop is needed because we sent no content length
    return true;
  }
  return false;
}
```

ricki-z, 2018-09-13:
The actual beta is changed so that iOS devices should also see a captive portal. But I can't test this. Download at: https://www.madavi.de/sensor/update/data/

ricki-z, 2019-03-14:
@LuchtwachtersDelft ist this issue solved? If yes please close the issue.

closed 2019-03-15 by LuchtwachtersDelft