opendata-stuttgart/sensors-software #256: BMP280 stopped working at the same time on multiple NodeMCUs

YNedderhoff, 2018-09-15:
I have 4 NodeMCUs with BMP280 sensors, some of them also have others. I store data in an InfluxDB instance on a Raspberry Pi. On Sept 4th at the same time data from all sensors stopped appearing in my database. Data from other sensors continues to be there.

I didn't do a lot of debugging, yet, but instead wanted to ask if there are any known issues, perhaps through a software update? The devices seem to run `NRZ-2018-109`. Or if that's not the case, if you have any debugging advice beyond restarting every involved device.

LuchtwachtersDelft, 2018-09-15:
Can confirm the Madavi.de graphs stopped at 00:00 UTC 5 Sept for my NodeMCU with a BME280, while my other NodeMCU with DHT22 kept going.

[image]

[image]

Check the CSV archive on that day. No data between 2018/09/04 23:58:31 and 2018/09/05 09:14:26.

https://www.madavi.de/sensor/csvfiles.php?sensor=esp8266-{Your NodeMCU ID}

Related? https://github.com/opendata-stuttgart/sensors-software/issues/255

YNedderhoff, 2018-09-15:
Thanks, that was the reason indeed.

closed 2018-09-15 by YNedderhoff