opendata-stuttgart/sensors-software #279: Sending data to external web server

Frickelpiet, 2018-10-18:
I want to send the sensors's data to an external web server. But sending the data is just working for one or two times after inserting the data at the sensors's configuration page. Has anybody an idea what could be the problem?

ricki-z, 2018-10-18:
Are you using a HTTPS connection?

Frickelpiet, 2018-10-18:
I'm not able to establish a connection via https (i.e. https://iot.xxxxx.de). My server is hosted by Strato and an ssl certificate is installed and works. Until now the script has created three datasets. In each case this has happened after I have changed the server address from for example https://iot.xxxxx.de to iot.xxxxx.de. But this behavior isn't reproducible. Sometimes it works, mostly not.

ricki-z, 2018-10-18:
The sensor only supports the ciphers RSA-SHA256 and RSA-SHA128. And certificates must be less or equal 2048 bits. A check showed me a length of 2048 bit for a Strato cert, but you should check this for your cert. Maybe that the ciphers are not supported by Strato.

ricki-z, 2018-10-18:
I have checked the Strato server that I have access to. They are configured to use ECDHE cipher only (perfect forward security). So this won't work with 'standard' Strato websites.

ricki-z, 2018-10-18:
Yes, the connection is only encrypted if you use port 443. If you set the port to 80 the connection will be unencrypted.
And please dont use 'http://' or 'https://' in the field 'Server'. This is only the server name. The protocol is set via the used port. So maybe it's working even on port 443 if you remove the 'https://'.

LuchtwachtersDelft, 2018-10-22:
I had a problem connecting to a LetsEncrypt site with ESP8266 Arduino Core 2.3.0, while I could connect to another LetsEncrypt site. That could depend on the server config. Updating to ESP8266 Arduino Core 2.4.2 worked for me.  
https://github.com/esp8266/Arduino/issues/2856

ricki-z, 2019-03-13:
@Frickelpiet can we close this issue?

closed 2019-04-28 by ricki-z